microsoft/psi #2: System.Runtime.InteropServices.COMException on initialization of visualizationService

bullbullet, 2018-04-11:
Hi, I am a member of Northwestern University QRG. We believed that we got the version from github, setup all paths correctly and built all projects successfully. However, when we started the WebcamWithAudioSample in Visual studio 2017 with x64 under Debug mode, we got the exception "System.Runtime.InteropServices.COMException: 'Retrieving the COM class factory for component with CLSID {8F2D3209-E9CF-49A0-BCCB-18264AC9F676} failed due to the following error: 80040154 Class not registered (Exception from HRESULT: 0x80040154 (REGDB_E_CLASSNOTREG)).'" 
on the 22 line of VisualizationClient.cs file: "this.visualizationService = (IRemoteVisualizationService)new VisualizationService();". We assume that we followed all instructions online and installed the right version of .Net 4.7. How can we solve this problem?

jdelliot, 2018-04-11:
We have attempted to make setting up Platform for Situated Intelligence Studio (PsiStudio) somewhat automatic. However, you must build and launch it at least one time before the client APIs will work.

- Have you already built and launched PsiStudio?
- Did you get prompted with an elevated permissions request for something called "Registry Console Tool"?
- Did you answer "Yes" to each time it prompted you?

The act of launching PsiStudio and clicking "Yes" to each of the prompts auto-registers the applications COM registry entries.

bullbullet, 2018-04-11:
Thanks for the help. It works now.

closed 2018-04-12 by jdelliot