microsoft/psi #4: Kinect v1 support

willieNU, 2018-06-19:
Is the Kinect v1 supported?  I've been using the Kinect v2 with psi, running the sample, and that all works fine.  Since we have a few extra v1s lying around, I was hoping to run things with those.  Before I devote too much time into this effort, I was wondering if the Kinect v1 is even likely to work with psi.

dparkar, 2018-06-19:
If by V1 you mean from the one that came with XBox 360, then I am interested in this as well.

jdelliot, 2018-06-20:
Presently, Kinect v1 is not supported. Kinect v1 would work just fine should with Psi by following the pattern laid down for Kinect v2. If you are interested in doing this work we would welcome a PR back and provide guidance and feedback along the way.

dparkar, 2018-06-22:
@willieNU From where did you get the USB adapter for the Kinect V1 ?

dparkar, 2018-07-02:
nvm ... I ordered it from Amazon.

closed 2018-07-09 by jdelliot