microsoft/psi #9: Building Runtime/Microsoft.Psi in Ubuntu

dreichCSL, 2018-06-28:
Using Ubuntu 18.04. I'm trying to build Sources/Runtime/Microsoft.Psi as part of one of the provided samples and am getting an error because my "ilasm" tool isn't generating the debug file when compiling the provided MemoryAccess.il library (it is generating the MemoryAcess.dll file, though). The code that compiles the library file can be found at the bottom of: 
Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj 

Is there something obvious that I'm missing here?

AshleyF, 2018-06-28:
Trying to diagnose this. If you could tell me the `ilasm --version` you're using, that could help (I have 5.10.0.140). Also, if you could try this: `ilasm IL/MemoryAccess.il /dll /output=./test.dll /DEBUG=OPT`. In particular the `/DEBUG=OPT` should emit a `.mdb`. Does it for you?

Thanks in advance. I'll continue to try to repro here...

dreichCSL, 2018-06-29:
The ilasm code I was using previously was part of coreCLR. I've installed the latest mono version now. ilasm --version here is 5.12.0.226, using an older version (4. something) gave me a different error (couldn't even compile the .il file). Anyway, with the latest mono/ilasm version everything's working fine. Thanks for quick response!

closed 2018-06-29 by dreichCSL