microsoft/psi #11: Ubunutu Installation problems

bakuzen, 2018-07-02:
I am attempting to install this on Ubuntu 16.04. 

After cloning psi, installing .NET core and VS Code, I tried to run build.sh but there was a missing ilasm binary necessary for building. 

I tried this: https://github.com/dotnet/coreclr/blob/master/Documentation/building/linux-instructions.md which, after compilation, generates the ilasm binary, but if I try to run it by pointing my PATH env var to it, I get segfaults. 

(This is Casey Kennington, btw)

AshleyF, 2018-07-03:
Hi Casey,

Although \psi is built on .NET Standard, we do still depend on the `ilasm` tool from Mono. This is missing from our setup docs. I'll add this.

Please try [installing the latest Mono](https://www.mono-project.com/download/stable/#download-lin) and using that version of `ilasm`.

bakuzen, 2018-07-05:
Hi AshleyF, 

That worked for me (at least, `./build.sh` processed successfully). I tried mono before, but it threw some errors. Following the link you provided apparently there was a more up-to-date mono version that did work.

closed 2018-07-09 by jdelliot

AshleyF, 2018-07-09:
Good to hear. I added this to [the build instructions](https://microsoft.github.io/psi/BuildingPsi).
Thanks for reporting this!

bakuzen, 2018-08-01:
Hi again, 

I was able to get the Bing ASR example working, but not without some effort. This feedback might be useful:

*  after cloning `psi`, installing .NET core and VS Code, I tried to run `build.sh` but there was a missing `ilasm` executable 
  * the latest version of mono devel's `ilasm` works: https://www.mono-project.com/download/stable/#download-lin
  * I needed to install scriptcs https://www.strathweb.com/2015/11/running-c-scripts-and-snippets-in-visual-studio-code-with-scriptcs/
  * the trick is to use `dotnet` in the console in a folder while coding in the VS Code editor https://code.visualstudio.com/docs/other/dotnet
  * In the end, I didn't use the code, I used the nuget packages. Adding nuget packages: https://stackoverflow.com/questions/40675162/install-a-nuget-package-in-visual-studio-code
  * I needed to set the following in the .csproj file:

```
<PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp2.0</TargetFramework>
    <RuntimeIdentifier>ubuntu.16.04-x64</RuntimeIdentifier>
  </PropertyGroup>
```
before getting all of the Linux-ready nuget packages for psi: https://microsoft.github.io/psi/NuGetPackagesList