microsoft/psi #13: Unable to install Microsoft.Psi.Visualization.Windows

willieNU, 2018-07-17:
I'm trying to install the new package (0.4.216.2-beta), but it is unable to resolve a dependency.  Output of trying to install is this:

Attempting to gather dependency information for package 'Microsoft.Psi.Visualization.Windows.0.4.216.2-beta' with respect to project 'KioskMain', targeting '.NETFramework,Version=v4.7'
Gathering dependency information took 469.6 ms
Attempting to resolve dependencies for package 'Microsoft.Psi.Visualization.Windows.0.4.216.2-beta' with DependencyBehavior 'Lowest'
Unable to resolve dependency 'Microsoft.Psi.Data'. Source(s) used: 'nuget.org', 'Microsoft Visual Studio Offline Packages'.
Time Elapsed: 00:00:00.4755751
========== Finished ==========

chitsaw, 2018-07-18:
Thanks for reporting this issue. Just wanted to let you know that we have hotfixed the issue and are awaiting approval to get the missing package published to nuget.org.

chitsaw, 2018-07-18:
The package Microsoft.Psi.Data 0.4.216.3-beta has been published on NuGet.org.

closed 2018-07-18 by chitsaw