microsoft/psi #17: Resizing Kinect depth image causes error

admayber, 2018-11-08:
The following code block generates a "generic GDI+ error":

`            using (Pipeline pipeline = Pipeline.Create("PsiNucStorage"))
            {
                KinectSensorConfiguration kinectSensorConfig = new KinectSensorConfiguration();
                kinectSensorConfig.OutputColor = true;
                kinectSensorConfig.OutputDepth = true;
                KinectSensor kinectSensor = new KinectSensor(pipeline, kinectSensorConfig);

                Exporter store = Store.Create(pipeline, "psiKinect", @".\data");

                Emitter<Shared<EncodedImage>> rgbImageStream = kinectSensor.ColorImage.Resize(100, 100).EncodeJpeg(100, DeliveryPolicy.LatestMessage).Out;
                Store.Write(rgbImageStream, "rgbImages", store, true, DeliveryPolicy.LatestMessage);

                Emitter<Shared<EncodedImage>> depthImageStream = kinectSensor.DepthImage.Resize(100, 100).EncodePng(DeliveryPolicy.LatestMessage).Out;
                Store.Write(depthImageStream, "depthImages", store, true, DeliveryPolicy.LatestMessage);

                Console.WriteLine("Starting data collect");
                pipeline.Run();
            }`

The culprit seems to be the resize operation on the depth image stream - removing that stage resolves the issue.

chitsaw, 2018-11-08:
Try the following patch on Sources\Imaging\Microsoft.Psi.Imaging\ImagePool.cs:
```
diff --git a/Sources/Imaging/Microsoft.Psi.Imaging/ImagePool.cs b/Sources/Imaging/Microsoft.Psi.Imaging/ImagePool.cs
index fcd9f449f..f5a20ed38 100644
--- a/Sources/Imaging/Microsoft.Psi.Imaging/ImagePool.cs
+++ b/Sources/Imaging/Microsoft.Psi.Imaging/ImagePool.cs
@@ -17,10 +17,10 @@ namespace Microsoft.Psi.Imaging
         /// <param name="width">Width of image requested</param>
         /// <param name="height">Height of image requested</param>
         /// <param name="pixelFormat">Pixel format for requested image</param>
-        /// <returns>Returns an image frmo the pool</returns>
+        /// <returns>Returns an image from the pool</returns>
         public static Shared<Image> GetOrCreate(int width, int height, PixelFormat pixelFormat)
         {
-            return KeyedSharedPool<Image, ValueTuple<long, PixelFormat>>.GetOrCreate(ValueTuple.Create((long)(width * height), pixelFormat), () => Image.Create(width, height, pixelFormat));
+            return KeyedSharedPool<Image, ValueTuple<int, int, PixelFormat>>.GetOrCreate(ValueTuple.Create(width, height, pixelFormat), () => Image.Create(width, height, pixelFormat));
         }

         /// <summary>
```

admayber, 2018-11-08:
Error still occurs, unfortunately. In case it's helpful, the error is originating at line 216 of Microsoft.Psi.Imaging\Operators.cs, in the middle of this code block:

```
graphics.ScaleTransform(xScale, yScale);
graphics.DrawImage(image.ToManagedImage(), new Point(0, 0));   \\ line that causes error
return ImagePool.GetOrCreate(bitmap);
```

chitsaw, 2018-11-08:
Thanks for the additional info. We'll investigate further and get back to you.

chitsaw, 2018-11-08:
The Resize operator does not work with the raw depth image format. In order to perform imaging operations on the depth stream, it should first be converted to a color or grayscale image like so:
```csharp
var depthImageStream = kinectSensor.DepthImage.ToGray().Resize(100, 100).EncodeJpeg(90, DeliveryPolicy.LatestMessage).Out;
```
We should probably throw a more helpful exception than the underlying generic GDI+ error.

admayber, 2018-11-08:
Ahh okay, that corrects the issue for me. Thank you!

closed 2018-11-08 by admayber