microsoft/psi #18: Can't visualize data stored using Kinect sample

admayber, 2018-11-09:
When I attempt to use Studio to open the stores saved out while running the Kinect sample, I cannot visualize any of the saved streams. Any 2D data streams (i.e., video) open a visualization panel with no data, and attempting to plot any of the 1D streams causes the program to crash.

Just to confirm that I'm doing this correctly, here's the process:

-Run the Kinect sample for some length of time
-Stop the program and close the Psi Studio window
-Run Psi Studio
-Open the catalog file of the store that was created when running the Kinect sample
-Select "plot" (resulting in a crash) or "visualize" (resulting in an empty visualization panel) on any of the streams in the store

admayber, 2018-11-09:
Yes to both, window appears and I can see the video stream just fine.

chitsaw, 2018-11-09:
It turns out the pipeline is not being shutdown properly on exit, resulting in the data store metadata getting corrupted. This is likely why you are unable to visualize any streams in the store. The data is still there, and we will be adding a feature to PsiStudio in the next release to automatically repair the store. We should also resolve this issue in the KinectSample. In the meantime, I would suggest applying the following patch to KinectSample\Program.cs:
```
diff --git a/Samples/KinectSample/Program.cs b/Samples/KinectSample/Program.cs
index 79d9d908d..bb4b075ff 100644
--- a/Samples/KinectSample/Program.cs
+++ b/Samples/KinectSample/Program.cs
@@ -82,7 +82,7 @@ namespace MultiModalSpeechDetection
                 convertedAudio.PipeTo(speechDetector);
 
                 // Use the Kinect's face track to determine if the mouth is opened
-                var mouthOpenAsFloat = kinectFaceDetector.Faces.Select((List<Microsoft.Psi.Kinect.Face.KinectFace> list) =>
+                var mouthOpenAsFloat = kinectFaceDetector.Faces.Where(list => list.Count > 0).Select((List<Microsoft.Psi.Kinect.Face.KinectFace> list) =>
                 {
                     if (!detected)
                     {
@@ -113,7 +113,7 @@ namespace MultiModalSpeechDetection
 
                 // Create a stream of landmarks (points) from the face detector
                 var facePoints = new List<Tuple<System.Windows.Point, string>>();
-                var landmarks = kinectFaceDetector.Faces.Select((List<Microsoft.Psi.Kinect.Face.KinectFace> list) =>
+                var landmarks = kinectFaceDetector.Faces.Where(list => list.Count > 0).Select((List<Microsoft.Psi.Kinect.Face.KinectFace> list) =>
                 {
                     facePoints.Clear();
                     System.Windows.Point pt1 = new System.Windows.Point(
@@ -173,7 +173,10 @@ namespace MultiModalSpeechDetection
                 ptsVis.Configuration.Radius = 3;
                 ptsVis.Configuration.XMax = Width;
                 ptsVis.Configuration.YMax = Height;
-                pipeline.Run();
+
+                pipeline.RunAsync();
+                Console.WriteLine("Press any key to exit...");
+                Console.ReadKey(true);
             }
         }
     }
```

admayber, 2018-11-09:
Ah okay, I was wondering if that might be the issue but wasn't sure how to do a graceful shutdown of the pipeline. This fixed the issue, thanks!

closed 2018-11-09 by admayber