microsoft/psi #23: Can't get AudioCapture to work on Linux

amussell, 2019-02-22:
I am running Ubuntu 16.04.3 on a Dell XPS 13 2015

I am trying to use the AudioCapture component and I am getting the following error.
```
Unhandled Exception: System.DllNotFoundException: Unable to load shared library 'asound' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libasound: cannot open shared object file: No such file or directory
   at Microsoft.Psi.Audio.LinuxAudioInterop.Open(Void** handle, String name, Int32 capture, Int32 mode)
   at Microsoft.Psi.Audio.LinuxAudioInterop.Open(String name, Mode mode, Int32 rate, Int32 channels, Format format, Access access)
   at Microsoft.Psi.Audio.AudioCapture.OnPipelineStart()
   at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action)
   at Microsoft.Psi.Scheduling.Scheduler.Run(Object workItem)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
```
I made sure that ALSA was installed and up to date. How do I get this library?

AshleyF, 2019-02-22:
Please try `apt install libasound2-dev` to get the library.

amussell, 2019-02-22:
That works! Is this expected for setup?

closed 2019-02-22 by amussell

AshleyF, 2019-02-22:
Excellent. Yes, we need to add that to the documentation.
Thanks for reporting!