microsoft/psi #28: How to use Diagnostics

willieNU, 2019-07-18:
I remember seeing a demo of the Diagnostics, but I cannot remember how to recreate it.  And I can't find any documentation on this yet.  I have found the API documentation for things in the Diagnostics namespace, but that hasn't helped me in figuring out how to use it.

I'm currently not interested in visualizing this data in PsiStudio.  I want to have a component that can read the diagnostics data.  I will then have a component with a websocket allowing me to access this realtime data through a web interface.  I am primarily interested in verifying that all components are operational.  But once I have the connection all working, I'll probably have other diagnostics information available via the web interface.

AshleyF, 2019-07-19:
Hi Willie,

You're right, we're missing documentation on this new feature. I'll work on this.

To get you started, `Pipeline.Create(...)` takes optional `enableDiagnostics` and `diagnosticsInterval` parameters. Setting `enableDiagnostics: true` will begin streaming `PipelineDiagnostics` messages on the `myPipeline.Diagnostics` emitter of the created `Pipeline`. These are snapshots of the complete graph at 100ms intervals (or an interval of your choice `diagnosticsInterval: TimeSpan.FromMilliseconds(10)`).

Each `PipelineDiagnostics` message contains the whole graph of `PipelineElements` and `Subpipelines`. `PipelineElements` contain diagnostics information about all of the `Emitters` and `Receivers`. The latter of which contain lots of statistics about throttling, queuing, processed/dropped counts, latencies, etc. This should be plenty of information to author a "watchdog" verifying that all your components are operational.

If ever you _do_ want to visualize in PsiStudio. You can write this `Diagnostics` stream to the store, open in the tool and visualize just as any other stream.

```
var store = Store.Create(myPipeline, "MyStore", "C:\Data");
Store.Write(myPipeline.Diagnostics, "MyDiagnosticsStream", store);
```

Hope this helps!

chitsaw, 2020-09-17:
Diagnostics are now documented [here](https://github.com/microsoft/psi/wiki/Diagnostics)

closed 2020-09-17 by chitsaw