microsoft/psi #32: Subscriber handler for other types

cagdask, 2019-10-24:
I'm playing around rosturtlesim code and I wanted to change subscription to another datatype. I assume I dont have to write a new message type for an existing message type in ros (i.e., sesors.range) but how about the event handler method signature? How should PoseUpdate(IEnumerable<Tuple<string, RosMessage.RosFieldVal>> position) should change in that case?

AshleyF, 2019-10-24:
You're correct that you shouldn't have to create a new message type for `sensor_msg/Range` it's ready-made as `RosMessageTypes.Sensor.Range.Def` [here](https://github.com/microsoft/psi/blob/master/Sources/Integrations/ROS/Microsoft.Psi.ROS/RosMessageTypes.fs#L91-L118). For example:

```
this.rangeSubscriber = this.node.Subscribe(RosMessageTypes.Sensor.Range.Def, RangeTopic, this.RangeUpdate);
```

Where `RangeTopic` is the topic name and `RangeUpdate` is your handler for the ROS messages; converting to events, for example, with something like:

```
public event EventHandler<float> RangeChanged;

private void RangeUpdate(IEnumerable<Tuple<string, RosMessage.RosFieldVal>> range)
{
    this?.RangeChanged(this, RosMessage.GetFloat32Val(range.First(x => x.Item1 == "range").Item2));
}
```

As you see, you need to know the field names (e.g. `"range"`). You can get this from the definition or [by looking it up here](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/Range.html), which will also tell you things like what the `"radiation_type"` may be (`ULTRASOUND=0, INFRARED=1`). The fields are just a sequence of name/`RosFIeldVal` pairs. An example of extracting multiple fields is [the Arm Control sample](https://github.com/microsoft/psi/blob/master/Samples/RosArmControlSample/UArm.cs#L146-L169).

Also, if you ever do need to construct the message def yourself, there's [an example in that sample as well](https://github.com/microsoft/psi/blob/master/Samples/RosArmControlSample/UArm.cs#L32-L49) and the main readme walks through how to get the MD5 hash and other bits you need to accomplish this.

Hopefully this help. Have fun!

cagdask, 2019-12-27:
it helps thank you

closed 2019-12-27 by cagdask