microsoft/psi #34: Trouble running simple app in Linux

angeliquetaylor, 2019-11-19:
I'm trying to run the simple app in Linux. I followed the build instructions but one thing that I was unsure about is whether I need nugget or not. The README page gives the impression that it is optional, "You can build applications based on Platform for Situated Intelligence either by leveraging nuget packages, or by cloning and building the code. Below are instructions:"

Can you confirm whether nugget is required or not?

Also, if nugget is required, which packages do I need from nuget.org

AshleyF, 2019-11-20:
Hello Angelique,

Which sample are you trying to run and what errors/issues are you seeing?

For example, the [`LinuxSpeechSample`](https://github.com/microsoft/psi/tree/master/Samples/LinuxSpeechSample) has _project references_ in the `.csproj` to `Microsoft.Psi.Audio.Linux` and `Microsoft.Psi.CognitiveServices.Speech`. Similarly for other samples. Having cloned the repo, you should be able to build from source like that — no NuGets required.

If instead you're developing your own stand-along app and would rather not have source-level dependencies, the built binaries are also published as NuGets (e.g. [`Microsoft.Psi.Audio.Linux`](https://www.nuget.org/packages/Microsoft.Psi.Audio.Linux/0.10.16.1-beta)) and can be added as package references to your project [in VS Code](https://marketplace.visualstudio.com/items?itemName=jmrog.vscode-nuget-package-manager) or [at the command-line](https://docs.microsoft.com/en-us/nuget/consume-packages/install-use-packages-dotnet-cli).

Hopefully that helps. Let us know how it goes.

angeliquetaylor, 2019-11-25:
Hi Ashley,

Sorry, I have to back track a bit because  tried resolving my prior issue and now the psi system won't build. I'm getting the following error on Linux, Ubuntu 16.04

g++: error: /libavdevice/libavdevice.so: No such file or directory
g++: error: /libavfilter/libavfilter.so: No such file or directory
g++: error: /libswresample/libswresample.so: No such file or directory
g++: error: /libavcodec/libavcodec.so: No such file or directory
g++: error: /libavformat/libavformat.so: No such file or directory
g++: error: /libavutil/libavutil.so: No such file or directory
g++: error: /libswscale/libswscale.so: No such file or directory
make: *** [Microsoft.Psi.Media.Native.so] Error 1
Failed to load ��, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load �, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load ��, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load ��n, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load ���, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load ��, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load �_Y, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load �?^, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load ��, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load �, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
./build.sh: line 14: cd: ./Sources/Integrations/ROS/Microsoft.ROS/: No such file or directory
Failed to load ���, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
Failed to load �, error: /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so: cannot open shared object file: No such file or directory
The library libhostfxr.so was found, but loading it from /usr/share/dotnet/host/fxr/5.0.0-alpha.1.19564.1/libhostfxr.so failed
[... cut ...]
  - Installing .NET Core prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063370
./build.sh: line 24: cd: ./Samples/PsiRosTurtleSample/: No such file or directory

Any ideas why I'm getting this error? I tried adding the path to the .so files to /etc/ld.so.conf and then I run sudo ldconfig but this did not resolve the issue.

Also, the installation link for .NET Core on Linux isn't working anymore: https://docs.microsoft.com/en-us/dotnet/core/linux-prerequisites?tabs=netcore2x

angeliquetaylor, 2019-11-30:
Hi Ashley,

I see that the Makefile uses environment variables FFMPEGLibDir. I installed ffmpeg from here: https://ffmpeg.org/download.html

However, I'm not sure what path to set FFMPEGLibDir to. Can you help me with this?

xiangzhi, 2019-12-03:
I just want to chip in since I'm also trying to get it working on our own Linux machines.

The way C#/Psi packages are built is very different from g++/cmake that are common on Linux systems. I was able to get them to build after I installed `dotnet-sdk-2.2` and `mono-devel`:
- https://docs.microsoft.com/en-us/dotnet/core/install/linux-package-manager-ubuntu-1604 (Make sure you install the 2.2 version and not 3.0 )
- https://www.mono-project.com/download/stable/#download-lin

Some packages should still fail to build since they have a optional requirements like FFMPEG, but you should see a bunch of `build succeeded`.  Another thing you can try is to navigate to the sample file location and build it yourself using the CLI commands. `dotnet build` and run it with `dotnet run`

If you aren't actively changing the runtime/components, I would stick with building applications using nugets. The intellisense (auto-complete) didn't work for me with the whole repo but works perfectly with nuget packages. Here's an example:
```
mkdir THIS_IS_MY_PROGRAM && cd THIS_IS_MY_PROGRAM
dotnet new console
dotnet add package Microsoft.Psi.Runtime --version 0.10.16.1-beta
```
Edit program.cs and add all the codes you want to run. To build and run it, just type `dotnet run`

angeliquetaylor, 2019-12-04:
Hi @xiangzhi! Thanks for your response. I followed the instructions to install dotnet-sdk-2.2 and mono-devel. However, I keep getting this error: ./build.sh: line 3: dotnet: command not found

Did you encounter this problem? Any idea what I might be missing?

xiangzhi, 2019-12-04:
> Hi @xiangzhi! Thanks for your response. I followed the instructions to install dotnet-sdk-2.2 and mono-devel. However, I keep getting this error: ./build.sh: line 3: dotnet: command not found
> 
> Did you encounter this problem? Any idea what I might be missing?

I'd try reopening the terminal. Could you also try to just type `dotnet` into the terminal and see what you get?

angeliquetaylor, 2019-12-04:
> I'd try reopening the terminal. Could you also try to just type `dotnet` into the terminal and see what you get?

I tried reopening a new terminal and just typing dotnet in the terminal but I still get the same error.  I look around to see what the issue is. Thanks!

xiangzhi, 2019-12-04:
> I tried reopening a new terminal and just typing dotnet in the terminal but I still get the same error. I look around to see what the issue is. Thanks!

My suspicion is that `dotnet-sdk` wasn't installed correctly. Good luck!

angeliquetaylor, 2019-12-04:
Hi forgot to set my environment variable. I added this to my bashrc:

export DOTNET_ROOT=$HOME/Documents/dotnet
export PATH=$PATH:$HOME/Documents/dotnet

Replace $HOME/Documents with the path to dotnet

The simple app works now. Thanks!

AshleyF, 2020-04-15:
Glad it works now. I'm closing this issue, but please feel free to reopen it if you're still having trouble.

closed 2020-04-15 by AshleyF