microsoft/psi #35: Store Creation Exception in Linux

xiangzhi, 2019-12-04:
When creating a PsiStore in Linux, the application fails and throws the following error:
```
Unhandled Exception: System.ObjectDisposedException: Cannot access a closed file.
   at System.IO.FileStream.Flush(Boolean flushToDisk)
   at System.IO.FileStream.get_SafeFileHandle()
   at System.IO.MemoryMappedFiles.MemoryMappedView.CreateView(SafeMemoryMappedFileHandle memMappedFileHandle, MemoryMappedFileAccess access, Int64 requestedOffset, Int64 requestedSize)
   at System.IO.MemoryMappedFiles.MemoryMappedFile.CreateViewAccessor(Int64 offset, Int64 size, MemoryMappedFileAccess access)
   at Microsoft.Psi.Persistence.InfiniteFileWriter.CreateNewExtent() in /home/xiangzht/Dev/psi/psi/Sources/Runtime/Microsoft.Psi/Persistence/InfiniteFileWriter.cs:line 294
   at Microsoft.Psi.Persistence.InfiniteFileWriter..ctor(String path, String fileName, Int32 extentSize, Boolean append) in /home/xiangzht/Dev/psi/psi/Sources/Runtime/Microsoft.Psi/Persistence/InfiniteFileWriter.cs:line 52
   at Microsoft.Psi.Persistence.StoreWriter..ctor(String name, String path, Boolean createSubdirectory, Boolean append) in /home/xiangzht/Dev/psi/psi/Sources/Runtime/Microsoft.Psi/Persistence/StoreWriter.cs:line 89
   at Microsoft.Psi.Data.Exporter..ctor(Pipeline pipeline, String name, String path, Boolean createSubdirectory, KnownSerializers serializers) in /home/xiangzht/Dev/psi/psi/Sources/Runtime/Microsoft.Psi/Data/Exporter.cs:line 44
   at Microsoft.Psi.Store.Create(Pipeline pipeline, String name, String rootPath, Boolean createSubdirectory, KnownSerializers serializers) in /home/xiangzht/Dev/psi/psi/Sources/Runtime/Microsoft.Psi/Data/Store.cs:line 42
   at Microsoft.Psi.Samples.LinuxSpeechSample.Program.RunAzureSpeech() in /home/xiangzht/Dev/psi/psi/Samples/LinuxSpeechSample/Program.cs:line 88
   at Microsoft.Psi.Samples.LinuxSpeechSample.Program.Main() in /home/xiangzht/Dev/psi/psi/Samples/LinuxSpeechSample/Program.cs:line 49
```
The problem seems to be in `Sources/Runtime/Microsoft.Psi/Persistence/InfiniteFileWriter.cs` line 275. In Linux, the file is closed upon exiting the scope.
```
using (var file = File.Open(this.extentName, FileMode.Create, FileAccess.ReadWrite, FileShare.Read))
{
    newMMF = MemoryMappedFile.CreateFromFile(file, null, this.extentSize, MemoryMappedFileAccess.ReadWrite, HandleInheritability.Inheritable, false);
}
```
My current workaround is to drop the `using` and do the following instead.
```
var file = File.Open(this.extentName, FileMode.Create, FileAccess.ReadWrite, FileShare.Read);
newMMF = MemoryMappedFile.CreateFromFile(file, null, this.extentSize, MemoryMappedFileAccess.ReadWrite, HandleInheritability.Inheritable, false);
```
I'm not sure about the repercussions about not closing the file.

AshleyF, 2019-12-14:
I can repro this under Ubuntu. I'm not sure why not under Windows.

Removing the `using` looks like a good fix. However, I would change the last flag passed to `MemoryMappedFile.CreateFromFile(...)` to `true` allowing this to happen at the MMFs discretion; [according to the docs](https://docs.microsoft.com/en-us/dotnet/api/system.io.memorymappedfiles.memorymappedfile.createfromfile?view=netstandard-2.1#System_IO_MemoryMappedFiles_MemoryMappedFile_CreateFromFile_System_IO_FileStream_System_String_System_Int64_System_IO_MemoryMappedFiles_MemoryMappedFileAccess_System_IO_HandleInheritability_System_Boolean_), `leaveOpen` is "A value that indicates whether to close the source file stream when the MemoryMappedFile is disposed."

We'll include this fix in the next release. Thanks for finding!

xiangzhi, 2019-12-16:
Awesome, thanks! Looking forward to the next release.

closed 2019-12-16 by xiangzhi