microsoft/psi #37: Connect with the webcam through "rtsp protocol"?

HaogangBao, 2019-12-12:
I am trying to connect a webcam through "rtsp protocol". I found in the "webcam sample", It uses the camera integrated into the computer,.Should I  use the "Mediacapture" component or any other components to find another webcam? If it is the "Mediacapture" component, which place I can modify to find the rtsp address?
Thank you very much for answering!

chitsaw, 2019-12-13:
Hi - the MediaCapture component only works with integrated or physically-connected cameras. We don't currently have a component that captures rtsp media streams, but it should be possible to build one,

chitsaw, 2019-12-18:
Closing issue. Feel free to reopen if necessary.

closed 2019-12-18 by chitsaw