microsoft/psi #38: PsiStudio Unhandled Exception (IOException: Cannot locate resource 'themes/images/resize_grip.png'.)

xiangzhi, 2019-12-13:
In Debug mode, PsiStudio will sometimes throw the following exception:
```
IOException: Cannot locate resource 'themes/images/resize_grip.png'.
```
It is safe to ignore the exception (click continue and PsiStudio still works.) but it will be nice if its handled, or the reference to image taken away.

chitsaw, 2019-12-19:
[posted on behalf of @sjdent]

Hi Xiangzhi

This is a strange error.  We do create a brush from this resource in Common.xaml, but it looks like we’re not using it anywhere.  I’m confused that you only sometimes get this error, but not all the time.  The image path (themes/images/resize_grip.png) is incorrect, it should be just (/images/resize_grip.png), probably because the image path is using a simple relative URI and not a pack URI.

There are several bitmap images in that file as well that would suffer from the same issue, I believe they’re not currently being used either.  If you change the URI at Common.xaml Line 34 to “pack://application:,,,/Images/resize_grip.png” the problem should go away for you, but I’d still like to get to the bottom of why you sometimes see this error.  Please let me know if you manage to create a good repro for this error.

xiangzhi, 2019-12-27:
I think I have a method to recreate the exception but I think its a very contrived case and related to my other issue #39. 

When I'm writing the raw Kinect 2 Image to the store and tried to visualize it live with PsiStudio, it will throw this exception upon attempts to visualize the image stream. I would say this is contrived since its unwise to both write and attempt to visualize raw image stream.s

chitsaw, 2020-04-15:
Resource paths have been updated in dc11ff3f87fccf139f2c3e21fc97a8da4c1d57f5. Closing issue.

closed 2020-04-15 by chitsaw