microsoft/psi #40: RealSense: System.BadImageFormatException

victorywys, 2019-12-13:
I'm trying to use the Intel Realsense cameras in Psi. I downloaded the SDK 2.0 and added the environment variable. The building process is successful. When I changed the WebcamWithAudioSample to a RealsenseSample, I got the following exception:
```
System.BadImageFormatException: Could not load file or assembly 'Microsoft.Psi.RealSense_Interop.Windows.x64, Version=0.10.16.1, Culture=neutral, PublicKeyToken=null'. An attempt was made to load a program with an incorrect format.
File name: 'Microsoft.Psi.RealSense_Interop.Windows.x64, Version=0.10.16.1, Culture=neutral, PublicKeyToken=null'
   at Microsoft.Psi.RealSense.Windows.RealSenseSensor.Start(Action`1 notifyCompletionTime)
   at Microsoft.Psi.Executive.PipelineElement.<Activate>b__47_0() in C:\Users\thisiswys\Downloads\psi-master\Sources\Runtime\Microsoft.Psi\Executive\PipelineElement.cs:line 259
   at Microsoft.Psi.Executive.PipelineElement.<>c__DisplayClass42_0.<TrackStateObjectOnContext>b__0() in C:\Users\thisiswys\Downloads\psi-master\Sources\Runtime\Microsoft.Psi\Executive\PipelineElement.cs:line 143
   at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action, SchedulerContext context) in C:\Users\thisiswys\Downloads\psi-master\Sources\Runtime\Microsoft.Psi\Scheduling\Scheduler.cs:line 525
 ```

Here are the lines I changed in the sample:
[image]

I know that BadImageFormatException is always related to a mismatch of the target platform and  #7 has encountered a similar problem. I analyzed my Microsoft.Psi.RealSense_Interop.Windows.x64.dll and believe that this file is in x64 format. And I ran the sample in x64 mode as follows:
[image]

but still I'm getting the exception. What could be the problem?

chitsaw, 2020-01-13:
We tried to repro your problem but were unable to. Could we have you change your active solution platform (the second dropdown in the second image from your previous post) from "x64" to "AnyCPU". The original Psi.sln only contains an "AnyCPU" platform configuration in which all project platform targets are pre-configured correctly.

Once you've done that, please right-click on the Microsoft.Psi.RealSense.Windows.x64 project in the Solution Explorer, select Properties, and verify that the Platform target under the Build tab is set to x64 and the Prefer 32-bit checkbox is grayed out as shown below:

[image]

Do this also for the RealSenseSample project (i.e. the sample project which you added).

Next, right-click on the Microsoft.Psi.RealSense_Interop.Windows.x64 project, select Properties, and verify that the Platform is set to x64 as shown below:

[image]

Let us know after you've done this if you still see the same issue. Thanks.

chitsaw, 2020-03-19:
Closing issue but please feel free to reopen if it is not resolved for you. Thanks.

closed 2020-03-19 by chitsaw