microsoft/psi #41: ROS third node connection

cagdask, 2019-12-23:
Hi I had to figure this one out from the hard way os I thought it will be nice to share. In order to connect to a node other than master or the other than dev machine on psi/Ros, we need to add its IP to RosDns, otherwise it will try to connect master node with same port for RequestTopic connection. This is mainly due to resolve code in RosDns :

`    let resolve rosMaster (uri: string) =
        let host = if uri.StartsWith("http://") then hostRegex.Match(uri).Groups.[1].Value else uri
        match Map.tryFind host known with
        | Some found -> uri.Replace(host, found)
        | None -> uri.Replace(host, rosMaster) // default`

I think it will be nice to document this somewhere in the tutorials. Otherwise it takes sometime to debug it, especially if you are not very familiar with F# like me.

AshleyF, 2020-01-07:
Thanks for noticing this! Apologies for the frustration. We'll add the following to the documentation:

Note that when running a distributed ROS system in which nodes are running on machines other than the ROS master, then hostname/IP mappings for these must be added (e.g. `RosDns.add("my-hostname", "10.1.2.3")`). Without such a mapping, the IP address of the master is assumed.

Does that make sense?

cagdask, 2020-01-10:
yeah that sounds good. Thank you AshleyF.

closed 2020-01-10 by cagdask