microsoft/psi #42: can Ros.RosMessageTypes.Sensor.Image definition be wrong?

cagdask, 2020-01-10:
Hi, I am not able to subscribe image messages on psi by using RosMessageTypes.Sensor.Image definition. I am not 100% sure whether the problem is on the psi side it might be on the publisher side as well. However I have minimum control on the publisher and I was able to fix the problem on psi by using a custom message definition as follows : 

`        private RosMessage.MessageDef depthImageMessageDef = RosMessage.CreateMessageDef(
           "sensor_msgs/Image",
           "060021388200f6f0f447d0fcd9c64743",
           new[]
           {
                        Tuple.Create("header", RosMessage.RosFieldDef.NewStructDef(Standard.Header.Def.Fields)),
                        Tuple.Create("height", RosMessage.RosFieldDef.UInt32Def),
                        Tuple.Create("width", RosMessage.RosFieldDef.UInt32Def),
                        Tuple.Create("encoding", RosMessage.RosFieldDef.StringDef),
                        Tuple.Create("is_bigendian", RosMessage.RosFieldDef.BoolDef),
                        Tuple.Create("step", RosMessage.RosFieldDef.UInt32Def),
                        Tuple.Create("data", RosMessage.RosFieldDef.NewVariableArrayDef(RosMessage.RosFieldDef.UInt8Def)),
           });`

The only difference I see is that on uint32 fields are defined as int and data field is defined as byte array at Microsoft.Ros.RosMessageTypes.Sensor.Image. 

By the way, I am able to subscribe other types of messages such as : RosMessageTypes.Geometry.Twist.Def and RosMessageTypes.Sensor.Range.Def

AshleyF, 2020-01-13:
I believe the problem is indeed on the \psi side. We [do have the correct field definitions (UInt32 width/height and UInt8[] data)](https://github.com/microsoft/psi/blob/master/Sources/Integrations/ROS/Microsoft.Psi.ROS/RosMessageTypes.fs#L166-L172), _but_ we are missing the MD5 hash! This has been fixed now. Thanks so much for pointing it out and sorry for the trouble.

closed 2020-01-13 by AshleyF

cagdask, 2020-01-13:
no problem at all and thank you for the fix.