microsoft/psi #43: interop in linux :feature request:

bakuzen, 2020-02-12:
Interop works beautifully when the psi app is running in Windows (and I can interop with Linux modules in other languages) using zmq. However, I'd like to setup an app on a Linux box and it would be really helpful to have interop capabilities.

AshleyF, 2020-02-13:
What specific issue are you running into? In the latest release, interop (`Microsoft.Psi.Interop`) as well as the core \psi runtime are .NET Standard libraries and _are_ expected to work cross-platform (any mixture of Linux/Mac/Windows). Please let us know of roadblocks we can investigate.

bakuzen, 2020-02-13:
It might be a version problem. Here's my command and the response:

`>dotnet add package Microsoft.Psi.Interop
  Writing /tmp/tmpFbCiBD.tmp
info : Adding PackageReference for package 'Microsoft.Psi.Interop' into project '/home/casey/psi_logger/psi_logger.csproj'.
info : Restoring packages for /home/casey/psi_logger/psi_logger.csproj...
info :   GET https://api.nuget.org/v3-flatcontainer/microsoft.psi.interop/index.json
info :   OK https://api.nuget.org/v3-flatcontainer/microsoft.psi.interop/index.json 229ms
error: Unable to find a stable package Microsoft.Psi.Interop with version 
error:   - Found 5 version(s) in nuget.org [ Nearest version: 0.10.16.1-beta ]
error: Package 'Microsoft.Psi.Interop' is incompatible with 'all' frameworks in project '/home/casey/psi_logger/psi_logger.csproj'.
`

AshleyF, 2020-02-13:
Very strange! I can repro this. However, if I append a specific version it works: `dotnet add package Microsoft.Psi.Interop -v 0.10.16.0-beta`.

AshleyF, 2020-02-13:
BTW, I just verified that the following works with both ends running under Ubuntu. Hopefully this works for you:
```python
import zmq, json

socket = zmq.Context().socket(zmq.SUB)
socket.connect("tcp://localhost:12345")
socket.setsockopt(zmq.SUBSCRIBE, '')

while True:
   [topic, message] = socket.recv_multipart()
   j = json.loads(message)
   print "Message: ", repr(j['message'])
```
```csharp
using System;
using Microsoft.Psi;
using Microsoft.Psi.Interop;
using Microsoft.Psi.Interop.Format;
using Microsoft.Psi.Interop.Serialization;
using Microsoft.Psi.Interop.Transport;

namespace Test
{
    class Program
    {
        static void Main(string[] args)
        {
            using (var p = Pipeline.Create())
            {
	        var gen = Generators.Range(p, 0, 1000, TimeSpan.FromMilliseconds(100));
                var sin = gen.Select(x => Math.Sin(x / 100.0));
                var mq = new NetMQWriter<double>(p, "sin-topic", "tcp://localhost:12345", JsonFormat.Instance);
                sin.PipeTo(mq);
                p.Run();
            }
        }
    }
}
```

bakuzen, 2020-02-13:
That worked! Thank you!

(though I needed to use the same version for Microsoft.Psi.Runtime)

AshleyF, 2020-04-15:
Glad it worked. I'm closing this issue, but please feel free to reopen it if you're still having trouble.

closed 2020-04-15 by AshleyF