microsoft/psi #44: Running PsiStudio results in Kinect Exception

burghto911, 2020-03-03:
I've been trying to get PsiStudio up and running and have run into the same issue as issue #5 .  The Psi.Sln built without issue however, after I installed the Kinect SDK, I still can't seem to get past the issue #5 and get the same exceptions.  I'm trying to Debug Microsoft.Psi.PsiStudio for Any CPU.  Here are the exceptions:
1. BadImageFormatException: Could not load file or assembly 'Microsoft.Kinect, Version=2.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies. Reference assemblies should not be loaded for execution.  They can only be loaded in the Reflection-only loader context. (Exception from HRESULT: 0x80131058)
2. BadImageFormatException: Cannot load a reference assembly for execution.

Looks like the Microsoft.Kinect dll is found in the C:\Users\LaptopName\.nuget\packages\microsoft.kinect\2.0.1410.19000\lib\net45 location.  

Should Psi point to the Kinect install location C:\Program Files\Microsoft SDKs\Kinect\v2.0_1409\Assemblies location?  They look like the exact same dll only in different locations.

Any help is appreciated!

burghto911, 2020-03-03:
I see that the Microsoft.Psi.Kinect.Windows Dependencies have System dll's from the "Program Files (x86)" which typically means these are 32-bit dlls.  Could this be an issue and/or how do I point to x64 system dlls?

chitsaw, 2020-03-03:
Sorry that you're running into this issue. We are trying to reproduce it and get you a solution. As to your question, there shouldn't be any need to modify any of the system dependencies in Microsoft.Psi.Kinect.Windows.

Have you tried rebooting your machine after installing the Kinect SDK? It is possible that the required DLLs are not being picked up from the Global Assembly Cache after the install without a reboot.

burghto911, 2020-03-03:
Yes I have rebooted my machine and reinstalled the Kinect SDK a few times.  I did have some trouble installing the Kinect SDK (Error 1920) but was able to finally install it by installing from the Program Data->Package Cache.  Not sure if this is causing some of the issues?

I check the the Global Assembly Cache and Microsoft.Kinect is not in there.  How would I install it in the GAC?

chitsaw, 2020-03-03:
You should not have to install Microsoft.Kinect in the GAC - the installer takes care of that. Though you do say that you had trouble installing using the installer, so there could have been some dependencies that did not get installed properly. I'm not sure that installing from the ProgramData\PackageCache necessarily does the right thing if the original install from the Setup exe did not complete without errors.

Regardless, I just realized we have been referring above to the Kinect SDK, not the Kinect _Runtime_ (which is what is needed in order to run PsiStudio). Could you verify that you are installing the package from this location: https://www.microsoft.com/en-us/download/confirmation.aspx?id=44559 ?

The solution and link to Kinect SDK in #5 no longer applies as this was an old requirement. The updated build instructions in the [wiki](https://github.com/microsoft/psi/wiki/Building-the-Codebase) points to the correct package you need to install (which I have provided the [link](https://www.microsoft.com/en-us/download/confirmation.aspx?id=44559) to here as well).

burghto911, 2020-03-03:
I had tried installing the Kinect runtime previously also but got an Error 1920 which I looks like a permissions issue.  The error occurs when I run the installer with and without admin privileges.  When I run the install manually from the Package Cache, it fails on starting up the Kinect Monitor service.  When I look in the services control panel during the install, the Kinect Monitor is disabled.  After rebooting, the Kinect Monitor service is not present - probably not a good thing :)

I looked around on some of the forums to resolve the Error 1920 and there was some discussion about being able to log on as a service so I added my user under the "Log on as a service" group policy but this didn't seem to help the install problem.

chitsaw, 2020-03-04:
Have you tried the solution mentioned here: https://social.msdn.microsoft.com/Forums/en-US/c8bc9c7b-7bdc-4b51-9fa8-f4962a82f533/installation-of-the-kinect-for-windows-sdk-20-fails-on-windows-81 ?

Other than that I am not sure what else to suggest as it does look like it is an issue you are having with the Kinect Runtime installation. I would also make sure you uninstall everything Kinect-related (SDK, Runtime) that you previously installed, then do a clean reboot before again attempting to install the [Kinect Runtime](https://www.microsoft.com/en-us/download/confirmation.aspx?id=44559).

burghto911, 2020-03-04:
I have not tried this approach but now I have and that was it!!  Unknowingly, I had an older "Visual C++ Redistribution" package installed and after removing this the Kinect Runtime install worked perfectly!  The Kinect Monitor Service is up and running and the PsiStudio runs as well!!  

Thanks so much for your help.  This issue is closed - phew!

closed 2020-03-04 by burghto911