microsoft/psi #45: Documentation update: Brief Introduction Offline Visualisation

DeCatNiels, 2020-03-04:
This is just a small documentation update that should probably be added.
To be able to run the **brief introduction example: 5.1 Offline Visualization.**
this one --> https://github.com/microsoft/psi/wiki/Brief-Introduction#51-offline-visualization
These Nuget references where missing for me when I was following this introduction:

- Microsoft.Psi.Audio
- Microsoft.Psi.Audio.Windows
- Microsoft.Psi.Imaging
- Microsoft.Psi.Speech

chitsaw, 2020-03-06:
Thank you for reporting this. You're right - Section 5.1 of the tutorial should begin with a new .NET Framework project (with 'Prefer 32-bit' unchecked). We'll update this.

chitsaw, 2020-04-15:
Updated [the documentation](https://github.com/microsoft/psi/wiki/Brief-Introduction#51-offline-visualization).

closed 2020-04-15 by chitsaw