microsoft/psi #46: InvalidOperationException in Generator, Creates Zombie Component and Blocks Pipeline with no indications.

xiangzhi, 2020-03-10:
Here is an example to illustrate my problem:
```
            using (var pipeline = Pipeline.Create())
            {
                var sameTime = DateTime.Now;
                var fakeList = new List<(int, DateTime)>();
                fakeList.Add((0, sameTime));
                fakeList.Add((1, sameTime));

                // Create a generator
                var gen = Generators.Sequence(pipeline, fakeList);
                gen.Do(x => Console.WriteLine("Hello World"));

                pipeline.Run();
                Console.WriteLine("Done!");
                Console.ReadLine();
            }
```
The expected output would be output of "Hello World" twice before the pipeline shuts down. However, if you run this code, it will output "Hello World" once and nothing else. The program will keep running without any outputs forever.

The problem is that the two elements have the same originating time and the generator's emitter has thrown a ``Attempted to post a message without strictly increasing originating time or that is out of order in wall-clock time`` exception and the pipeline has start shutting down but fail to shutdown the generator completely. 

After diving deeper into the code of `Generator{T}.cs` and `Generators.cs`, I hypothesize this is the sequence of errors that cause the pipeline to hang.

1. `Generator{T}.cs` line 68 throws the InvalidOperationException because the two elements have the same originating time.
1. `Generator.cs` line 108 never returns and the `this.nextMessageTime` is never updated.
1. The exception is caught by the pipeline and start shutting all the components down with the final originating time to be AFTER the last good message in the generator.
1. The pipeline's `DeactivateComponents` method calls the  `Stop` method of the `generator.cs` (line 76).
1. `this.nextMessageTime` will be the old time and is before the `this.finalMessageTime` and won't completely stop the component, thinking there are more messages from it.
1. The pipeline thinks the generator is still catching up and generating old messages and wait for it.
1. It waits forever......

I can think of some fixes like catching the error on this level or additional checks in the Stop method. I was wondering whether y'all had faced similar problems and if this bug is already fixed in an internal release. 

My actual use case is when converting ROS Messages to Psi Store, sometimes the messages were published so fast they have the exact same publish time. We used a generator to output all the messages and they stop working suddenly.

chitsaw, 2020-03-11:
Indeed! This is a bug that we have fixed and will be in the next release which we are planning to push out in a week or so.

xiangzhi, 2020-03-11:
Awesome, looking forward to the next release! 
I'll keep this open until the release just in case someone else has the same problem.

chitsaw, 2020-04-15:
Fixed in dc11ff3f87fccf139f2c3e21fc97a8da4c1d57f5

closed 2020-04-15 by chitsaw