microsoft/psi #47: kinect requirement

sstoyanchev, 2020-03-18:
Hi, I am just getting started with \psi and going over the  'BriefIntroduction' Tutorial.

On step 5 it says: 'Kinect for Windows SDK is required to run PsiStudio'

Does it actually require kinect device or only SDK? Would it work with a regular computer camera/microphon?

thank you
-Svetlana

danbohus, 2020-03-18:
It does not require a Kinect device, it currently requires just the SDK. We will aim to remove that dependency in a future release.

sstoyanchev, 2020-03-18:
Great, thanks. Are there any other SDKs that it will require? (I need to ask an admin to install them and would help to do all at once)

danbohus, 2020-03-19:
The required and optional prerequisites for building the codebase are described here: https://github.com/microsoft/psi/wiki/Building-the-Codebase

Hope this helps, but let us know if you run into additional issues.

danbohus, 2020-04-09:
I assume this is resolved, so closing this issue, but feel free to reopen in case you need more assistance.

closed 2020-04-09 by danbohus