microsoft/psi #48: Reading structured data in from NetMQ

willieNU, 2020-04-03:
I am setting up NetMQ/ZeroMQ to communicate between psi and python.  I have some structured data going to python and getting parsed in python just fine.  For example, the message payload of `{ 'help' : '1' }` is easily read into a python dictionary.

The other direction has been harder.  I'm trying to respond with the following message (full message just for completeness):
`{'message': {'say': 'hello'}, 'originatingTime': '2020-04-02T23:24:30.010592'}`
In psi, I wasn't able to setup a NetMQSource<Dictionary<string,string>> to emit a message with this payload.  I changed the NetMQSource type to dynamic, and it emits fine, and I can see it is an ExpandoObject (which is expected based on my reading of JsonFormat.cs).  But now I'm not sure what to do with that.  Maybe I'm just not familiar with how to use ExpandoObject.  Is there any way to get this to a Dictionary?

AshleyF, 2020-04-04:
Hey Willie,

You're right that the correct way to receive a structured message on the \psi side is as a `dynamic`. You should be able to unpack fields simply by knowing the names (e.g. 'message.say'). Or you can treat the underlying 'ExpandoObject' as a dictionary by merely casting it: 'IDictionary<string, object> dict = message;` If you really want to convert it to some structured type on the .NET side (or even to a dictionary of strong types), you'd do that with a `.Select(...)` unpacking the dynamic and constructing a strong type of your own. It's not automatic.

Hope that helps!

AshleyF, 2020-04-15:
Again, I hope that helped. I'm closing this issue, but please feel free to reopen it if you're still having trouble.

closed 2020-04-15 by AshleyF

willieNU, 2020-04-15:
Ashley, yes this was helpful.  Here's the code I ended up using:

`var mqs = new NetMQSource<dynamic>(this, "message-topic", "tcp://127.0.0.1:12002", JsonFormat.Instance);
mqs.Select(msg => ((IDictionary<String, Object>)msg)).PipeTo(Out);`

I'm working on a video tutorial of how all this works, and I'll share that video.