microsoft/psi #51: Deserializing error, possibly related to enums

willieNU, 2020-05-26:
Replaying data from a data store has generally been very useful, and this is the first time I have had a type that was not able to be deserialized.  It looks like the error is related to an enum nested inside of the class.  What is the best way to handle this?  Just move the enum to outside the class?  Do I need to define a deserializer? Or maybe I need to store with primitive types and reconstruct the type after deserializing?

System.AggregateException
  HResult=0x80131500
  Message=Pipeline 'default' was terminated because of one or more unexpected errors (Failed to create a deserializer for type System.Collections.Generic.EnumEqualityComparer`1[[EmotionRepresentation.EmotionCategory+Category, EmotionRepresentation, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]], mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 because no type was registered for this name and the source type System.Collections.Generic.EnumEqualityComparer`1[[EmotionRepresentation.EmotionCategory+Category, EmotionRepresentation, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]], mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 could not be found. Add a reference to the assembly containing this type, or register an alternate type for this name.)
  Source=Microsoft.Psi
  StackTrace:
   at Microsoft.Psi.Pipeline.ThrowIfError()
   at MainCore.ReasonerTest.Main(String[] args) in C:\linux\repos\temp\psiAssistiveAgent\MainCore\ReasonerTest.cs:line 44

Inner Exception 1:
SerializationException: Failed to create a deserializer for type System.Collections.Generic.EnumEqualityComparer`1[[EmotionRepresentation.EmotionCategory+Category, EmotionRepresentation, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]], mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 because no type was registered for this name and the source type System.Collections.Generic.EnumEqualityComparer`1[[EmotionRepresentation.EmotionCategory+Category, EmotionRepresentation, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]], mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 could not be found. Add a reference to the assembly containing this type, or register an alternate type for this name.

chitsaw, 2020-05-26:
Sometimes we get these sorts of exceptions during deserialization when \psi is unable to locate the `Type` needed to deserialize an object (which may be nested).  Typically this happens when you are trying to deserialize an object which contains a polymorphic object via an interface or base class reference, and the runtime is not able to instantiate a serializer for the derived type because it does have a reference to the defining assembly.

The way to get around this is to register the missing type for serialization somewhere in your code before you attempt to read the data from the store. Based on your exception message above, could you try adding the following to your code:

```
using EmotionRepresentation;
using Microsoft.Psi.Serialization;
...
KnownSerializers.Default.Register<EmotionCategory.Category>();
```

willieNU, 2020-05-26:
Unfortunately that did not work.  Any other suggestions?

chitsaw, 2020-05-27:
You could try registering with the full type name, though I would have thought this to be unnecessary unless your assembly/type name has changed in some way:

```
KnownSerializers.Default.Register<EmotionCategory.Category>("EmotionRepresentation.EmotionCategory+Category, EmotionRepresentation, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null");
```

Other than that, I can't think of anything else at the moment. If you would be willing to share your code and the data store, perhaps I could take a closer look to get to the bottom of this (it would be helpful for us too to understand why this doesn't work).

willieNU, 2020-06-03:
Yes, I will send some code and a data store next week.