microsoft/psi #52: Build dependencies

colombod, 2020-05-28:
When building I see that there are dependencies on older sdk versions
[image]

The solution has a lot of dependencies, could this be simplified with a solution for 

- all
- runtime (cross platform)
- additional modules (speech and all other things like audio and so on)
- platform specific modules

That could improve the experience for contributors and code exploration.

chitsaw, 2020-05-29:
Thanks for the suggestion! This is something we will look into for a future release. In the nearer term, our next release will move all C++ projects to Visual Studio 2019, which will resolve the missing toolset dependency you encountered above.

colombod, 2020-06-03:
I am interested in contributing to the portable part of the framework (see #53 ) but the all build requirements and one big solution makes it hard.

I am using the PSI to create a framework for the [piTop](https://www.pi-top.com/products/pi-top-4) platform, this is the current state of my [github project](https://github.com/colombod/pi-top)