microsoft/psi #53: stream operator : moving average, sliding window

colombod, 2020-06-03:
I see there is a window operator but that is waiting for the window to be filled, or at least it seems that if I do ```Average(TimeSpan.FromSeconds(1))``` it will cause the producer to emit every second instead of retaining the inbound frequency.

This is useful to computer rms, loudness and smoothing signal without loosing frequency. Another application would be to create signal filters to smooth curves.

A common thing here would be to provide lowpass, gates, compressors, kalman, 1 euro operators

willieNU, 2020-06-05:
I think the Windowing operators do emit with the same frequency as they receive.  However, this is only true after the window has filled.  One the window is full, every new received data point will create a new emitted one. 

I have also found that `Average(TimeSpan.FromSeconds(1))` and `Average(10)` create quite different outputs.  For the former, it outputs an average even with the first data point, though it is delayed by a second.  For the latter, which is also delayed a second if the input stream is once every 100 ms,  the first average emitted is the average of the first 10 data points.

I've tested this with the following code:
```
var sequence = Generators.Sequence(p, 0d, x => x + 0.1, 100, TimeSpan.FromMilliseconds(100));
var sin = sequence.Select(t => Math.Sin(t));
sin.Do(t => Console.WriteLine($"Sin: {t}"));
sin.Average(TimeSpan.FromSeconds(1)).Do(t => Console.WriteLine($"Avg: {t}"));
//sin.Average(10).Do(t => Console.WriteLine($"Avg: {t}"));
p.Run();
```

With the code as is, this outputs the following:
```
Sin: 0
Sin: 0.0998334166468282
Sin: 0.198669330795061
Sin: 0.29552020666134
Sin: 0.389418342308651
Sin: 0.479425538604203
Sin: 0.564642473395035
Sin: 0.644217687237691
Sin: 0.717356090899523
Sin: 0.783326909627483
Sin: 0.841470984807896
Sin: 0.891207360061435
Avg: 0
Avg: 0.0499167083234141
Sin: 0.932039085967226
Avg: 0.0995009158139631
```

If I uncomment that one line and comment out the line above, I get the following:
```
Sin: 0
Sin: 0.0998334166468282
Sin: 0.198669330795061
Sin: 0.29552020666134
Sin: 0.389418342308651
Sin: 0.479425538604203
Sin: 0.564642473395035
Sin: 0.644217687237691
Sin: 0.717356090899523
Sin: 0.783326909627483
Avg: 0.417240999617581
Sin: 0.841470984807896
Avg: 0.501388098098371
```

I'm not sure if either of these is what you are looking for.

colombod, 2020-06-05:
Strange, I use a sequence generator to read from a sensor 10 a second. If I use average or window on that stream the frequency seems to be the one with the window size.

colombod, 2020-06-05:
Still I am not sure how to emit a sliding window over a stream of size t1 with t2 overlap.

colombod, 2020-06-05:
this is the program I am writing [Program.cs](https://github.com/colombod/pi-top/blob/master/src/PiTop.PsiApp/Program.cs)

But it seems do not getting more than 7-8 samples per seconds from the sensor, it is running on a [pi-top](https://www.pi-top.com/products/pi-top-4)  device

xiangzhi, 2020-06-05:
I tested a similar code on my machine. Like @willieNU mentioned, the average is only posted one second later and only consist of the first value. I believe this is a bug in the Timespan version of Average (`Average(TimeSpan.FromSeconds(1))`).  There are two mysteries, (1) one second delay and (2) only consisting of the first value.

I peeked into the code (`psi/Sources/Runtime/Microsoft.Psi/Components/RelativeTimeWindow.cs`) and found this:
```
  protected override void Receive(TInput value, Envelope envelope)
        {
            // clone and add the new message
            var message = new Message<TInput>(value, envelope).DeepClone(this.recycler);

            // time-based
            if (this.initialBuffer)
            {
                var clone = this.buffer.DeepClone();
                this.buffer.Enqueue(message);
                if (this.CheckRemoval())
                {
                    this.initialBuffer = false;
                    this.ProcessWindow(clone, false, this.Out);
                    this.ProcessRemoval();
                    this.ProcessWindow(this.buffer, false, this.Out);
                }
            }
            else
            {
                this.buffer.Enqueue(message);
                this.ProcessRemoval();
                this.ProcessWindow(this.buffer, false, this.Out);
            }
        }
```
`this.ProcessWindow` is the function that takes the current window, does the required operation, and post it. Looking at the code, we can see it only publish it when it checks for removal. This explains the one second delay, since it only removed stuff from the buffer after a second. This solve the first mystery. 

The second mystery is a bit more complex. Psi is a very message driven pipeline. The average will only be posted on the frequency it receives an input. In `this.processWindow` it checks which was the oldest non-served message and use it as an anchor point to calculate the average with that message as the MAX point of the window. This explains mystery two since there isn't any older message than the first message.

A quick fix for both problem would be to change the function to:
```
        protected override void Receive(TInput value, Envelope envelope)
        {
            // clone and add the new message
            var message = new Message<TInput>(value, envelope).DeepClone(this.recycler);

            // time-based
            if (this.initialBuffer)
            {
                var clone = this.buffer.DeepClone();
                this.buffer.Enqueue(message);
                if (this.CheckRemoval())
                {
                    this.initialBuffer = false;
                    this.ProcessWindow(clone, false, this.Out);
                    this.ProcessRemoval();
                }
                else
                {
                    this.ProcessWindow(this.buffer, false, this.Out);
                }
            }
            else
            {
                this.buffer.Enqueue(message);
                this.ProcessRemoval();
                this.ProcessWindow(this.buffer, false, this.Out);
            }
        }
```
I believe its a design choice where the messages are handled in order even if the buffer has newer messages since one of Psi's goal is reproducibility. This is the output from Willie's code after the fix
 ```
Sin: 0
Sin: 0.09983341664682815
Avg: 0
Sin: 0.19866933079506122
Avg: 0.04991670832341408
Sin: 0.2955202066613396
Avg: 0.09950091581396313
Sin: 0.3894183423086505
Avg: 0.14850573852580723
Sin: 0.479425538604203
Avg: 0.1966882592823759
Sin: 0.5646424733950354
Avg: 0.24381113916934707
Sin: 0.644217687237691
Avg: 0.28964418691587396
Sin: 0.7173560908995227
Avg: 0.3339658744561011
Sin: 0.7833269096274833
Avg: 0.376564787394259
Sin: 0.8414709848078964
```

Anyway, I'm curious to hear what are the Psi team's thoughs on this.

danbohus, 2020-06-05:
Thanks @colombod for raising the issue and @willieNU and @xiangzhi for your responses. We've been looking into this issue, so let me start by first adding a couple of clarifications. 

Re: Average(), @willieNU and @xiangzhi are correct on both accounts: 

1) the operator keeps the cadence of the source stream, but only "once the window is full" (more below)
2) the two operators Average(TimeSpan) and Average(Int) do behave differently. 

Let me start with the second point. The reason the two flavors of `Average()` behave differently is that the two operators reflect two fundamentally different ways of performing windowing over streams: with windows defined by time, or with windows defined by a message counter. The Average(), as well as other mathematical and statistical windowing operators we have rely underneath on a more general [Window operator](https://github.com/microsoft/psi/wiki/Windowing-Operators) which also has these two flavors. The difference in behavior @willieNU is observing is because when windowing over time the contract basically is: "for every message on the source stream, produce an output message containing an array with all messages in the specified relative time window, anchored in that message", whereas when windowing over message counts, the contract is: "for every message, produce an array with a number of messages in front or behind it". Note that you can do a variety of relative time windows, such as 

Window([-500ms, 0]) - the last 500 ms
Window([-500ms, +200ms]) - an asymmetric window with 500ms in the past and 200ms in the future, relative to current message
Window([-500ms, -100ms]) - a window that actually does not include the current message. 

and similarly for windows over message counters

Window([-5, 0]) - the last 6 messages, including the current one
Window([-5, -1]) - the previous 5 messages, etc. 

Now, `Average(TimeSpan.FromSeconds(1))` is a shortcut for Average([-1s, 0]), so "for every message give me all the messages in the last second". So when the operator sees the first message on the stream, it knows there are no other messages before it, so it produces a window that contains only that message. In contrast, when the `Average(10)` operator sees the first message, it cannot yet output "the 9 messages before this one + this one" because there are no such messages before it. A window containing "9 past messages + this one" cannot be constructed at this point. `Average(10)` will actually not output any buffer corresponding to messages 0, 1, 2, ... 8. Once it sees 9, it can form a window of size 10, so only then it starts outputting. (As an aside, this investigation has led us to consider/discuss whether we should add a version of message counter indexing that allows for clipped windows to be output at the beginning and end of the stream, but that does not currently exist). 

Now, getting back to 1) above, it is true that the current default implementation of the Window operator (which average relies on) waits at the beginning in wall clock time to output the buffer corresponding to that first message until the window is full. After that it keeps the cadence of the source. @willieNU has already provided an example, and you can also verify this like below. If I do: 

```
Generators.Repeat(p, 0, TimeSpan.FromMilliseconds(50))
    .Average(TimeSpan.FromSeconds(2))
    .Do((m, e) => Console.WriteLine($"@{e.OriginatingTime.TimeOfDay}"));
```

Then, upon startup, there is a wait of two seconds, after which a line with the originating time is printed every 50 ms, on the original stream cadence (i.e. the window is sliding by 50 ms with every new message on the source stream). 

There is another overload of the Window operator that you can use if you want to not even wait at the beginning. That operator lets you specify a relative time interval that defines the window, a function that computes something over the messages, and a flag that indicates whether to wait for complete windows or post right away. So for instance, you can do I think what you are looking for by:

```
Generators.Repeat(p, 0, TimeSpan.FromMilliseconds(50))
    .Window(RelativeTimeInterval.Past(TimeSpan.FromSeconds(2)), messages => messages.Average(m => m.Data), waitForCompleteWindow: false)
    .Do((m, e) => Console.WriteLine($"@{e.OriginatingTime.TimeOfDay}"));
```

We are actually looking into whether this (no waiting) should be in fact the default behavior of `Window()` in the time domain (it shouldn't really wait, there's no reason to), and we might change that in a future release (stay tuned to this issue, we will keep it open, for more updates on that). Also, we plan to extend the [documentation for these windowing operators](https://github.com/microsoft/psi/wiki/Windowing-Operators) which is a bit sparse at this point. 

Hopefully the above helps provide some more clarity. Now, to help diagnose the issue @colombod is still seeing, can you give us a bit more detail? Here are a few questions:

- Are you aiming to do windowing in the time domain or in the message count domain (per above)? 
- Does your source keep a constant frequency in originating times, or does it have variable frequency? 
- Have you confirmed that the source gives the messages have the right rate (that you expect) before going into the Average operator?

[... cut ...] nderneath currently produce a buffer (when possible) for every message they see incoming. That means the anchor of the buffer is driven by the incoming messages, and the window advances by one message at a time, with a cadence that is tied to the input. To clarify your question, are you trying to slide a window in the time domain by some specified constant _timespan-delta that is independent of what's in the source data stream_ and get the messages that the source stream contains in that window?

danbohus, 2020-06-26:
With the [latest release](https://github.com/microsoft/psi/releases/tag/v0.12.53.2-beta), we have addressed the issue of the delay induced by the Window operator discussed above. Also, we have recently expanded the documentation on [Windowing Operators](https://github.com/microsoft/psi/wiki/Windowing-Operators), which now explains the various `Window()` operators in more depth, and describes how to construct a window that slides over a stream at a fixed cadence, independent of the cadence of the messages in the stream (see the Windows Defined Dynamically section in that document).

@colombod: let us know whether this has addressed your questions, or if there are any additional questions you have.

colombod, 2020-06-26:
@danbohus thank you all this looks pretty good and enough for me.

closed 2020-06-26 by colombod