microsoft/psi #54: Example or Guides or Tutorials for 3rd party visualizers in PsiStudio

xiangzhi, 2020-06-08:
Hi,

The latest release mentioned the addition of 3rd party visualizer support in PsiStudio. The release note included a small paragraph on how to add them. I was wondering if there is any documentation, guides or examples on how to create and use them? Any pointers on where to start will be helpful.

danbohus, 2020-06-09:
Unfortunately this feature is indeed not documented yet. We will work on adding a wiki page under "More Advanced Topics" on this matter. The next release, which should come out shortly, will also include an example of how to write and use a 3rd party visualizer that's external to PsiStudio. Stay tuned, we'll keep this issue opened, and post updates here as well as they arrive.

xiangzhi, 2020-06-09:
No problem, I'll dig through the code for now.