microsoft/psi #55: Building Azure Kinect Component on Linux/Ubuntu Machine.

xiangzhi, 2020-06-17:
I just downloaded the latest version of Psi and trying out the Azure Kinect Component. On my Ubuntu machine, it refused to build and throwing the following error. I built it using the `build.sh` script that came with Psi.
```
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  All projects are up-to-date for restore.
/home/xiangzht/.nuget/packages/microsoft.azure.kinect.sensor/1.3.0/build/netstandard2.0/Microsoft.Azure.Kinect.Sensor.targets(4,5): error : Azure Kinect only supports the x86/x64 platform ('AnyCPU' not supported) [/home/xiangzht/Dev/Psi/psi/Sources/Kinect/Microsoft.Psi.AzureKinect.x64/Microsoft.Psi.AzureKinect.x64.csproj]

Build FAILED.

/home/xiangzht/.nuget/packages/microsoft.azure.kinect.sensor/1.3.0/build/netstandard2.0/Microsoft.Azure.Kinect.Sensor.targets(4,5): error : Azure Kinect only supports the x86/x64 platform ('AnyCPU' not supported) [/home/xiangzht/Dev/Psi/psi/Sources/Kinect/Microsoft.Psi.AzureKinect.x64/Microsoft.Psi.AzureKinect.x64.csproj]
    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:00.49
```
I tried and successfully run it on my Windows Partition, so it seemed to be a Linux issue. I'm trying to debug it but any pointers will be helpful (Including if it won't work on Ubuntu)

xiangzhi, 2020-06-17:
I traced the problem to a typo in the platform declaration in the `.csproj` files. The SDK checks the `$(Platform)` property https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/release/1.3.x/src/csharp/k4a.props.

They check for the property `$(Platform)` but there is a typo in both `Sources/Kinect/Microsoft.Psi.AzureKinect.x64/Microsoft.Psi.AzureKinect.x64.csproj` and `Samples/AzureKinectSample/AzureKinectSample.csproj` that specifies the platform as `<Platforms>x64</Platforms>`. Changing both to `<Platform>x64</Platform>` solve the problem.

I don't know why this is a problem in Linux but not Windows.

xiangzhi, 2020-06-17:
I tested the Sample application and besides the body tracking, everything works correctly. 

If I include the body tracking code, I'll get the following error: 
```
/onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:97 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:91 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] CUDA failure 10: invalid device ordinal ; GPU=0 ; hostname=TBD-Beetle ; expr=cudaSetDevice(device_id_); 
Stacktrace:

Stacktrace:

[2020-06-17 17:02:46.889] [error] [t=9234] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/TrackerHost.cpp (157): Create(). /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:97 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:91 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] CUDA failure 10: invalid device ordinal ; GPU=0 ; hostname=TBD-Beetle ; expr=cudaSetDevice(device_id_); 
Stacktrace:

Stacktrace:

[2020-06-17 17:02:46.889] [error] [t=9234] [K4ABT] /home/vsts/work/1/s/src/sdk/k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```

It seemed to be an issue in the proprietary body tracking component. I'm using CUDA 10.2, and a search on the internet suggest it might be caused by different CUDA versions. I was able to use Azure Kinect's Body Tracking through the ROS driver, so it's probably not cause by the hardware.

chitsaw, 2020-06-19:
Thanks for reporting and investigating the build problem. $(Platforms) is actually a list of available platform configurations. The reason it worked on Windows and not on Linux is because the Visual Studio solution sets the Platform to x64. We will change it as you suggest.

As for the body tracking issue, not quite sure what is going on here. It looks as though you have already installed both the Azure Kinect [Sensor SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download) and [Body SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download).

Could you try running with body tracking in CPU mode by changing the `CpuOnlyMode` property in the `AzureKinectBodyTrackerConfiguration` to `true`?

xiangzhi, 2020-06-19:
No problem, thanks for looking into it! I'm still really unfamiliar with VS build process.

I have both SDK installed through the Ubuntu package manager. These are the packages that I have:
```
libk4a1.3/bionic,now 1.3.0 amd64 [installed,automatic]
libk4a1.3-dev/bionic,now 1.3.0 amd64 [installed]
libk4abt1.0/bionic,now 1.0.0 amd64 [installed,automatic]
libk4abt1.0-dev/bionic,now 1.0.0 amd64 [installed]
```
I tried both CPU & GPU versions and both throw the same error. The error is thrown at line 116 in `psi/Sources/Kinect/Microsoft.Psi.AzureKinect.x64/AzureKinectBodyTracker.cs` when the Tracker is created. I also doubled check that the configuration variables was propagated down correctly.

chitsaw, 2020-06-22:
Thanks for the feedback. We will follow up with the Azure Kinect team and get back to you.

chitsaw, 2020-08-02:
Reported the issue to the Azure Kinect team: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1304

KevinHwangAi, 2020-12-17:
I also had the same problem on Ubuntu 18.04. How is the problem solved now?

xiangzhi, 2020-12-17:
@KevinHwangAi I haven't tried the latest version, but I believe the body tracking component is still broken (It is caused by an external SDK).  The other parts of the component (rgb image, depth image, etc) works, just not the body tracker.

KevinHwangAi, 2020-12-17:
Now our development is on hold and we have to consider a new direction. I hope the Microsoft team can solve this problem as soon as possible.

sandrist, 2022-03-19:
Thanks for the reminder. It looks like there is a new version of the [Azure Kinect body tracker SDK](https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-download), and thankfully it is available via Nuget this time. We will update our \\psi component and test to see if it fixes the Linux issue. I'll follow up on this thread to let you know the results, but feel free to also try updating the package reference in a local fork/clone of the repo if you'd like to test it yourself in the meantime.