microsoft/psi #57: PsiStudio no longer able to visualize Kinect 2 Bodies.

xiangzhi, 2020-06-19:
After the newest update, PsiStudio no longer shows the option for visualizing the Kinect 2 Bodies. Any chance this is due to the spatial changes? This is with a new store that I collected with the most recent version of Psi.

danbohus, 2020-06-20:
We have recently moved the Kinect body visualizer out of PsiStudio into its own project, under `Microsoft.Psi.Kinect.Visualization.Windows` project under Sources\Kinect\Microsoft.Psi.Kinect.Visualization.Windows (this enabled us to remove the dependency PsiStudio had on the Kinect SDK). You can build this project and configure PsiStudio to load the visualizer from the external DLL by editing the PsiStudioSettings.xml file from your user's Documents\PsiStudio folder. 

To enable PsiStudio to load external DLLs, you can add an `<AdditionalAssemblies>` xml tag inside the main `<PsiStudioSettings>` tag in PsiStudioSettings.xml and specify the assembly you want to load, for instance:

```
<AdditionalAssemblies>C:\psi\Sources\Kinect\Microsoft.Psi.Kinect.Visualization.Windows\bin\Debug\net472\Microsoft.Psi.Kinect.Visualization.Windows.dll</AdditionalAssemblies>
```

The same holds true for the AzureKinectBody visualizer, which is available in the `Microsoft.Psi.AzureKinect.Visualization.Windows.x64` project under Sources\Kinect\Microsoft.Psi.AzureKinect.Visualization. If you want to add multiple external DLLs, use a single `<AdditionalAssemblies>` tag, and add them in a semicolon separated list, e.g.:

```
<AdditionalAssemblies>C:\psi\Sources\Kinect\Microsoft.Psi.Kinect.Visualization.Windows\bin\Debug\net472\Microsoft.Psi.Kinect.Visualization.Windows.dll;C:\psi\Sources\Kinect\Microsoft.Psi.AzureKinect.Visualization\bin\Debug\net472\Microsoft.Psi.AzureKinect.Visualization.Windows.x64.dll</AdditionalAssemblies>
```

We will document this more prominently shortly.

xiangzhi, 2020-06-20:
Thanks for the pointers Dan! I just added them and both Azure Kinect & Kinect 2 bodies are showing up now. Both are also good examples for me for writing my own visualizers.

closed 2020-06-20 by xiangzhi