microsoft/psi #58: Creating compounded custom ROS Message Scheme in C# Psi.

xiangzhi, 2020-06-22:
I'm trying to create a ROS Message Definition for the one of our custom ROS Message ([https://github.com/CMU-TBD/tbd_ros_msgs/blob/master/msg/HumanBody.msg](https://github.com/CMU-TBD/tbd_ros_msgs/blob/master/msg/HumanBody.msg)
) in Psi following the Psi ROS Examples.

The `RosMessage.CreateMessageDef` function takes in a `IEnumerable<Tuple<string, RosFieldDef>>` as the final argument. Since my message uses an existing message type: `geometry_msgs/Pose` instead of a field, I cannot add it to the `IEnumerable`.

My currently workaround is to add it straight into the F# code in `\psi\Sources\Integrations\ROS\Microsoft.Psi.ROS\RosMessageTypes.fs`.

I was wondering if there is a way to declare it in C# instead of messing with the F# code?

AshleyF, 2020-06-22:
It's pretty unwieldy in C#, but could you use `RosMessage.RosFieldDef.NewStructDef(RosMessageTypes.Geometry.Pose.Def.Fields)` in your custom definition?

xiangzhi, 2020-06-23:
Thanks for the tip! I haven't fully test it yet but I got it working now.

closed 2020-06-23 by xiangzhi