microsoft/psi #62: Exception when running Multiple Azure Kinect Body Trackers

xiangzhi, 2020-07-07:
I'm trying to run multiple azure kinect body trackers off a single machine. About 9 out of 10 times, I get the following exceptions at line 116@`psi\Sources\Kinect\Microsoft.Psi.AzureKinect.x64\AzureKinectBodyTracker.cs`
 ```
System.Runtime.InteropServices.SEHException
  HResult=0x80004005
  Message=External component has thrown an exception.
  Source=<Cannot evaluate the exception source>
  StackTrace:
<Cannot evaluate the exception stack trace>
```
The remaining one time works perfectly and I am able to collect the data. Any thoughts on how to debug this?  

Here's my test code:
```
           using (var p = Pipeline.Create(true))
            {
                var store = Store.Create(p, "test", @"C:\Data\Stores");

                var k4a1 = new AzureKinectSensor(p, new AzureKinectSensorConfiguration()
                {
                    OutputColor = true,
                    OutputInfrared = true,
                    DeviceIndex = 0,
                    BodyTrackerConfiguration = new AzureKinectBodyTrackerConfiguration()
                    {
                        CpuOnlyMode = false
                    }
                });

                var k4a2 = new AzureKinectSensor(p, new AzureKinectSensorConfiguration()
                {
                    OutputColor = true,
                    OutputInfrared = true,
                    DeviceIndex = 1,
                    BodyTrackerConfiguration = new AzureKinectBodyTrackerConfiguration()
                    {
                        CpuOnlyMode = false
                    }
                });

                k4a1.ColorImage.EncodeJpeg(90, DeliveryPolicy.LatestMessage).Write("cam1", store);
                k4a1.Bodies.Write("body1", store);
                k4a2.Bodies.Write("body2", store);
                k4a2.ColorImage.EncodeJpeg(90, DeliveryPolicy.LatestMessage).Write("cam2", store);

                p.Diagnostics.Write("diagnostic", store);
                p.RunAsync();
                Console.ReadLine();
            }
```

Here's the things that I tried with no success:
1. Having one of the body tracker use `CpuOnlyMode`
2. Run it in Release Mode

chitsaw, 2020-07-15:
It appears that this exception is being thrown from unmanaged code in the external Azure Kinect body tracking library. Unfortunately the E_FAIL HRESULT does not tell us much. You could try bumping the Azure Kinect log level as documented [here](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___logging.html).

Another possibility is that initializing multiple Azure Kinects simultaneously may be an issue, as [this comment](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1013#issuecomment-571638663) suggests. You could try adding a small delay just before the second call to `Tracker.Create` which threw the exception to see if that resolves it for you.

xiangzhi, 2020-07-15:
Thanks @chitsaw! The static lock fixed the problem. Since it is an easy enough fix, I just made a pull requst back to the main repo.

chitsaw, 2020-07-15:
Merged #64. Closing this issue.

closed 2020-07-15 by chitsaw