microsoft/psi #63: Automatic language detection in PSI

HaogangBao, 2020-07-13:
I am trying to use "Automatic language detection with the Speech SDK" of Aurze( https://docs.microsoft.com/en-us/azure/cognitive-services/speech-service/how-to-automatic-language-detection?pivots=programming-language-csharp ), but I found the language settings in the PSI is only support one language.  

How can I use the automatic language detection in the PSI？Thanks

chitsaw, 2020-07-14:
Automatic language detection is a relatively new feature that is only available through the Speech SDK. Unfortunately the built-in AzureSpeechRecognizer component in psi uses the WebSockets API to access the service, so this feature is not available at the present time via this component. We do plan to eventually migrate to the newer Speech SDK, but we do not have a firm timeline on when this will happen. In the meantime, you might try creating your own psi component that wraps the Azure Speech SDK client to expose just the functionality needed for your scenarios.

HaogangBao, 2020-07-14:
> Automatic language detection is a relatively new feature that is only available through the Speech SDK. Unfortunately the built-in AzureSpeechRecognizer component in psi uses the WebSockets API to [...]

Thanks, chitsaw. We would try to use this new feature in the PSI