microsoft/psi #67: Spectre-mitigated components installed, but still getting errors

annelo-msft, 2020-07-16:
Following the instructions in [Building the Codebase](https://github.com/microsoft/psi/wiki/Building-the-Codebase), I installed the Spectre-mitigated packages:

[image]

I'm still getting these errors:

[image]

Help?

chitsaw, 2020-07-16:
I think you may need to install v14.26 of the Spectre-mitigated libs instead since that is now the latest version. Could you try installing that and let us know if it works? If so we should also update the instructions :-)

annelo-msft, 2020-07-16:
Yes, that worked - it builds now!

[image]

Thank you so much!! :)

closed 2020-07-16 by annelo-msft