microsoft/psi #69: Unable to use AzureKinect nuget package in .net core WPF applications

KanaHayama, 2020-07-17:
It reports `Azure Kinect only supports the x86/x64 platform ('AnyCPU' not supported)`.
While, in Visual Studio, only `AnyCPU` is allowed to select for .net core WPF applications.
The target platform is actually set by `PlatformTarget` tag.

sandrist, 2020-07-17:
Setting `PlatformTarget` to be x64 is not enough, and unfortunately the steps you need to take in order to also change the "project platform" to x64 are a bit confusing in Visual Studio.

Take a look at the project properties for `AzureKinectSample` project to see what you need for your own new project:

[image]

To accomplish that, after you create the new .net core WPF app, you need to right-click on the solution and go to "Configuration Manager...". Find your project in the list and create a <New...> platform called "x64" using the dropdown menu. You can then also select <Edit...> from the dropdown menu and remove the AnyCPU platform.

[image]

Finally, right-click on the project in Visual Studio and select "Edit Project File" to make sure everything looks ok. You'll want the `Platforms` field to look like this:

[image]

KanaHayama, 2020-07-19:
Thank you.
With that `Create new solution platforms` unchecked, not I can create a `x64` platform target without an error popped out.

closed 2020-07-19 by KanaHayama