microsoft/psi #70: Visualize complex data with generics in psiStudio

TahsinMir, 2020-07-29:
I have been working with psi for research purposes where I am working with the speech recognizer. I wrote a separate component which takes in the output of the recognizer to do additional work on the data. While psiStudio can clearly visualize the recognizer(IStreamingSpeechRecognitionResult) data, latency and messages, it can only show latency and messages for my new component output which is of complex type containing lists and multiple generics. Is there any way to add new and complex data structures like this for psiStudio to visualize?

danbohus, 2020-07-31:
The framework and PsiStudio supports authoring and loading of 3rd party visualizers. Once authored (in your own dll/project), you can load and make them available in PsiStudio as described [on this page](https://github.com/microsoft/psi/wiki/3rd-Party-Visualizers).

Unfortunately, if you are interested in constructing a visualizer for a generic type, like `List<T>`, this is not currently supported. If you are interested however in building a visualizer for a particular type that uses generics, like `List<int>`, or `List<float>`, this is supported (for instance the [Points 3D visualization object](https://github.com/microsoft/psi/blob/master/Sources/Visualization/Microsoft.Psi.Visualization.Common.Windows/VisualizationObjects/Points3DVisualizationObject.cs) visualizes `List<Point3D>`. If the number of types `T` that you care about is known and relatively small, you can probably get around the limitation above by adding a visualizer for each of the `T`s that you care about, e.g. a visualizer for `List<int>` and a separate one for `List<float>`, etc.

Visualizers are implemented as various types of Visualization Objects -- you can find a lot of them in [this folder](https://github.com/microsoft/psi/tree/master/Sources/Visualization/Microsoft.Psi.Visualization.Common.Windows/VisualizationObjects), but each of them have an associated view. For instance, the [SpeechRecognitionVisualizationObject](https://github.com/microsoft/psi/blob/master/Sources/Visualization/Microsoft.Psi.Visualization.Common.Windows/VisualizationObjects/SpeechRecognitionVisualizationObject.cs) uses the [SpeechRecognitionVisualizationObjectView](https://github.com/microsoft/psi/blob/master/Sources/Visualization/Microsoft.Psi.Visualization.Common.Windows/Views/Visuals2D/SpeechRecognitionVisualizationObjectView.xaml). Constructing a generic view is where things get complicated. If you are interested in getting support for truly generic visualizers, we would recommend you open a separate issue, label it `feature request` and describe if possible a bit more the intended use case. 

We plan to add more documentation in the future on how to write visualizers, for now unfortunately you'll have to find and mirror the patterns for the existing visualizers. You can find them by looking for classes whose names end in VisualizationObject, and are decorated with a `VisualizationObject` attribute. If you are looking for one implemented outside of Visualization.Common, you can have a look at the [Azure Kinect body visualization object](https://github.com/microsoft/psi/blob/master/Sources/Kinect/Microsoft.Psi.AzureKinect.Visualization/AzureKinectBodyVisualizationObject.cs). There are several different patterns we have been using for building visualizers, often depending on their type (3D, 2D, timeline), but we plan to systematize the codebase of visualizers and converge on a few recommended patterns, which we will document more.

As a side note, just in case it ends up helping with anything in your scenario, the visualization infrastructure also supports something called stream adapters - you can find them by searching the codebase for the `[StreamAdapter]` attribute. These classes adapt streams from an input type to an output type. For instance, if you have a stream of type `T1`, a visualizer for streams of type `T2` and a stream adapter from `T1` to `T2`, PsiStudio will be able to use that adapter and visualizer you stream of `T1` as a stream of `T2`. These adapters can also be authored and loaded from 3rd party projects/dlls.

If you start writing a visualizer and run into specific problems / questions, let us know.

TahsinMir, 2020-07-31:
That was helpful. I was able to figure out what to do. Thank you

closed 2020-07-31 by TahsinMir