microsoft/psi #72: Setting Clock's replaySpeedFactor in Pipeline.Run/Pipeline.RunAsync through ReplayDescripter

xiangzhi, 2020-08-13:
I'm currently working on my components by replaying existing PsiStores and letting the data flow through my pipeline. As described in the wiki, there are currently two version of the code: **Real Time** vs **Maximum Speed**. Since I am interfacing with other systems, I was looking around for ways to change the speed without going unlimited. 

I found that the pipeline uses a `Clock` object which has a `replaySpeedFactor` that changes how fast the data is being replayed. Would it be possible to set that variable through the `ReplayDescripter`? Letting the application change how fast the messages are being replayed?

I was wondering whether such capabilities is on the roadmap? Is there a better way that I missed? I also have a working version with these changes and happy to push it back upstream.

chitsaw, 2020-08-15:
There were a number of issues that led us to only expose Real Time and Max Speed when replaying from stores. Chiefly among these was the confusing latency measurements that arise when you replay with clock enforcement *and* a speed factor that is not 1. Due to the time skew, one can no longer infer a message processing latency simply by comparing the originating time with the creation time of a message. 

We would like to understand your scenario which led you to implement a variable replay speed factor. Do you even care about the issue of output message latencies being incorrect?

xiangzhi, 2020-08-15:
My main use case is to have the Psi application process data and send to my ROS module. Here's an overview of what I'm trying to do (applies to the other two serialization problems too).

We are using multiple Azure Kinects to capture the scene around two robots. The data collection and processing of all the cameras are done by Psi and at the end of the pipeline, it is sent to ROS. ([https://github.com/CMU-TBD/psi/blob/tbd/visual-pipeline/main/TBD/TBD.Psi.VisualPipeline/CollectionPipeline.cs](https://github.com/CMU-TBD/psi/blob/tbd/visual-pipeline/main/TBD/TBD.Psi.VisualPipeline/CollectionPipeline.cs)).

I'm currently in the testing phase and have saved some PsiStore for replaying. The store saves the Azure Kinect outputs but before the merger/tracking components. During the replaying, I was hoping to speed up the playback to see how all the components works and how the ROS system reacts. Real Time takes a while to go through our data. Maximum Speed just dumps all the data at once & our external components would need to some how handle all those data. In our current tests, we are ignoring latencies but would like to maintain the same time difference/ratio between messages.

My main reason is probably to get some control over the speed of communication between Psi + ROS during playback. I also think this might be helpful in applications where you have to manually assess the output at runtime; You can save time by replaying the data faster once you determine latency isn't a problem.

chitsaw, 2020-08-15:
Thanks for the feedback - we will look into it. In the meantime I assume the solution which you have implemented works for now? And yes, exposing it through the `ReplayDescriptor` would be the best way to do it.

xiangzhi, 2020-08-15:
Yup, it works as intended. Thanks!

closed 2020-08-15 by xiangzhi