microsoft/psi #76: [Request] Exposing Azure Kinect Synchronization Options in Psi.

xiangzhi, 2020-08-25:
We are currently using multiple Azure Kinects to collect human data. Azure Kinect has a hardware sync feature through 3.5mm audio cables. 
Information: https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync
Example:  https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen

The Psi Azure Kinect Component currently does not expose those settings. I checked the underlying Azure Kinect CIL code and it does have the settings for master/subordinate control and we would need to build a wrapper around them. I envision it to be an abstracted wrapper that set certain device to be the master or subordinate. 

Is this feature on the roadmap? Again, I'm also happy to make those changes and send it back upstream since I might need them.

danbohus, 2020-08-27:
Certainly sounds like a useful feature. It's not on our roadmap currently, and we would welcome a PR if you end up enabling this scenario. 

Btw, I'm not sure I completely understand what you mean by "abstracted wrapper that set certain device to be master or subordinate". If this is a configuration setting that the underlying Azure Kinect API provides, is the idea to expand the configuration object for the corresponding Azure Kinect \psi component to expose that?

xiangzhi, 2020-08-27:
Cool! I'll see what I can do.

The SDK already provide the variables to set whether a sensor is the master or subordinate. I was thinking of exposing the enum that set the type of sensors [k4a_wired_sync_mode_t](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___enumerations_ga6404c655dd8656efb745c6ee90da2505.html#ga6404c655dd8656efb745c6ee90da2505) and all the delay variables that are needed to ensure the sensors don't interfere with each other. This will be done by adding them to the `AzureKinectSensorConfiguration`. I was also thinking of renaming the variables to be a bit more C#/Psi friendly.

closed 2020-08-27 by xiangzhi