microsoft/psi #78: How to creating Annotation Definition for Annotation Streams?

xiangzhi, 2020-09-01:
I'm testing the newly released annotation streams. When creating it, it asks me to pick a `annotation definition` but there are no options in it. 

[image]

Do I need to definite it separately in the code or the PsiStudioSettings.xml? PsiStudio crashes when I try to visualize the annotation stream and it seemed to be caused by a lack of annotation definition.

Thanks for pushing out this new feature! I'm very excited to use it in my research.

skdent, 2020-09-02:
Hi Xiangzhi

The new documentation for this feature isn't up yet, but here's what you need to do:

1) Create a new directory called <MyDocuments>\PsiStudio\AnnotationDefinitions
2) Drop the attached annotation definition file into this new directory.  This definition is a two-track annotation definition, one track is a finite/enumerated value, the other track is a free text value. NOTE: I needed to change the extension to ".txt" to add the file to this post; once you've copied it locally you need to change the extension back to ".pad"
3) Restart PsiStudio and you should now have an annotation definition you can test with.
4) Note that you should already have a dataset open when you create the annotation stream.
5) Once you've created the annotations stream and dragged it onto the visualization canvas, mark a selection on the timeline using SHIFT+Click and SHIFT+RightClick, and then RightClick and select "Add Annotation".
6) Once you have some annotations, you can select an annotation by clicking on it and then set its values in the Properties tab on the right of PsiStudio.
7) Note that the name of the stream will have an asterisk next to it if you've mad

e any edits to the annotation stream.  Save the store by RightClicking on the partition and selecting "Save Changes"

Hope that helps you out, let me know if you have any other questions.

[TestDefinition.txt](https://github.com/microsoft/psi/files/5164390/TestDefinition.txt)

xiangzhi, 2020-09-03:
Thanks, I just got it working on my computer! The annotation worked really well. 

One suggestion I have is that PsiStudio can throw some sort of error message if you try to add an annotation without selecting a portion of the timeline. Currently nothing happens which is weird.

closed 2020-09-03 by xiangzhi

skdent, 2020-09-03:
Thanks for the idea Xi, I'll implement that.  I'll also add some code to prevent users from creating an annotation stream without first selecting a definition.