microsoft/psi #83: Storing of Shared<EncodedImage> with different memory sizes will eventually all have the same sizes.

xiangzhi, 2020-09-06:
In a scenario such as the following:
```
k4a1.ColorImage.EncodeJpeg(quality: 10).Write("q10", store);
k4a1.ColorImage.EncodeJpeg(quality: 80).Write("q80", store);
```
We expected that the PsiStore stream `q10` will have a smaller message size than the `q80` stream because it is more. compressed. However, when we inspected it in PsiStudio, both has about the same average message size. 

We believe this problem comes from the `Shared<T> `infrastructure, where both streams share existing `EncodedImage` objects.  Because of `Shared<T>`, an existing `EncodedImage` object with a larger memorystream (previously used in the `EncodeJpeg(quality: 80)`) could be use in the smaller stream and that object, with its extended and unused memorystream will be encoded into PsiStore. This is caused by `EncodedImagePool.GetOrCreate` unable to distinguish between objects with different memory profiles.

chitsaw, 2020-09-11:
Thank you for reporting this and for the detailed analysis! We are currently working on a fix for this which will be coming soon.

chitsaw, 2020-09-16:
This has been fixed in 23b6cc2

closed 2020-09-16 by chitsaw