microsoft/psi #84: [Bug] NetMQSource from python, 'originatingTime' must be utc

dhgarcia, 2020-09-08:
Following the [Interop Transport Readme.md](https://github.com/microsoft/psi/blob/master/Sources/Runtime/Microsoft.Psi.Interop/Transport/Readme.md) example to produce messages from python:
```python
import zmq, random, datetime, json

context = zmq.Context()
socket = context.socket(zmq.PUB)
socket.bind('tcp://127.0.0.1:12345')

while True:
    payload = {}
    payload['message'] = random.uniform(0, 1)
    payload['originatingTime'] = datetime.datetime.now().isoformat()
    socket.send_multipart(['test-topic'.encode(), json.dumps(payload).encode('utf-8')])
```

This was not working to consume messages on the NetMQSource side. Nothing was being printed from the \Psi part.
Changing `datetime.now().isoformat()`  to `datetime.utcnow().isoformat()`  fix it.
Also works when using `useReceivedTimes = false`. So the publishing part was not the issue.

Is this true? Are the instructions wrong in terms of `payload['originatingTime'] = datetime.datetime.now().isoformat()` and `utcnow()` is required to generate the timestamps? 

using python 3.6.9 from Ubuntu 18.04, .NET core SDK 3.1.401

Thanks

AshleyF, 2020-09-09:
You're correct. The document should suggest using `utcnow()` rather than `now()`. Times in general in \psi are UTC.

I'm not able to actually repro a failure to read (in a simple pipeline like the below), but I can imagine how it could cause trouble when fusing streams with drastically different originating times or possibly outside of the range of the pipeline replay descriptor. At any rate, `utcnow()` should be used when emitting message that truly originate in Python.

```csharp
using (var p = Pipeline.Create())
{
    new NetMQSource<double>(p, "test-topic", "tcp://localhost:12345", JsonFormat.Instance)
        .Do((x, e) => Console.WriteLine($"Message: {x} ({e.OriginatingTime})"));
    p.Run();
}
```

We'll update the doc. Thanks much for pointing this out!

dhgarcia, 2020-09-09:
On my laptop `.now()` and `.utcnow()` differ by 1 hour. Maybe that's why?

Great! I'm glad, is an easy fix. I was puzzle for a while.
Thanks.

PR #85 (2020-09-09): Updated interop doc to suggest `datetime.utcnow()`
The document should suggest using utcnow() rather than now(). Times in general in \psi are UTC.

I'm not able to actually repro a failure to read (in a simple pipeline like the below), but I can imagine how it could cause trouble when fusing streams with drastically different originating times or possibly outside of the range of the pipeline replay descriptor. At any rate, utcnow() should be used when emitting message that truly originate in Python.

using (var p = Pipeline.Create())
{
    new NetMQSource<double>(p, "test-topic", "tcp://localh [... cut]
closed 2020-09-09 by AshleyF