microsoft/psi #86: Microsoft.Psi.Media.Linux  capture webcam with PixelFormatId.MJPEG

dhgarcia, 2020-09-09:
I'm trying to capture video from a webcam in Linux, as the [Python-Interop-Walkthrough](https://github.com/microsoft/psi/wiki/Python-Interop-Walkthrough) and the [LinuxWebcamWithAudioSample](https://github.com/Microsoft/psi-samples/tree/main/Samples/LinuxWebcamWithAudioSample)
But my camera only supports MJPEG.

```csharp
var mediaCapture = new MediaCapture(pipeline, 1280, 720, "/dev/video0", PixelFormatId.MJPEG);

var webcamFrames = mediaCapture.Out.Resize(640, 480);
webcamFrames.Do(_ => Console.Write('.'));
```
According to the [Microsoft.Psi.Media.Linux/MediaCapture.cs](https://github.com/microsoft/psi/blob/master/Sources/Media/Microsoft.Psi.Media.Linux/MediaCapture.cs) YUYV and MJPEG are supported.

```csharp
if (pixelFormat != PixelFormatId.BGR24 && pixelFormat != PixelFormatId.YUYV && pixelFormat != PixelFormatId.MJPEG)
            {
                throw new ArgumentException("Only YUYV and MJPEG are currently supported");
            }
```

But, this was not working. Nothing gets posted from the `.Out` stream. If I replace it with the `.Raw` there is an output.
Further exploration of the source suggest that MJPEG is not actually implemented, as it is not one of the options from  `public unsafe void Start(Action<DateTime> notifyCompletionTime)` https://github.com/microsoft/psi/blob/master/Sources/Media/Microsoft.Psi.Media.Linux/MediaCapture.cs#L135

```csharp
using (var sharedImage = ImagePool.GetOrCreate(this.configuration.Width, this.configuration.Height, PixelFormat.BGR_24bpp))
                    {
                        if (this.configuration.PixelFormat == PixelFormatId.BGR24)
                        {
                            // BGR code (lines 155-156)
                        }
                        else if (this.configuration.PixelFormat == PixelFormatId.YUYV)
                        {
                            // convert YUYV -> BGR24 code (lines 160-197)
                        }
                   }
```

What am I missing?
Is there a way to work directly with the `.Raw` stream?
How can I use the `Shared<byte[]>` and eventually transform that to an `Image` object?
Thanks!

AshleyF, 2020-09-16:
Thanks for reporting this. I believe you're correct that MJPEG is not fully supported. It is an allowed configuration and the `Raw` stream is indeed JPEG encoded `byte[]`s, but these are not decoded to `Shared<Image>` on the `Out` stream. We're looking into fixing this. I'll keep you posted.

dhgarcia, 2020-09-16:
Thanks.

In the meantime, Is there any help on how to work with the Raw stream? 
I couldn't find a way to properly use the Raw `Shared<byte[]>`

AshleyF, 2020-09-18:
The `MediaCapture` component now fully supports MJPEG. Thanks again for reporting this!

closed 2020-09-18 by AshleyF

dhgarcia, 2020-09-22:
Thanks for solving this issue!

In the [LinuxWebcamWithAudioSample](https://github.com/Microsoft/psi-samples/tree/main/Samples/LinuxWebcamWithAudioSample)  now the image is display with the wrong colors.
I assume BGR instead of RGB, and this is more a pixbuf issue, but I'm wondering if you know what can be done?
[line 108](https://github.com/microsoft/psi-samples/blob/544638a6ff161ba5b083891cd8325289e25e3115/Samples/LinuxWebcamWithAudioSample/MainWindow.cs#L108)

Thanks.

AshleyF, 2020-09-23:
I can't seem to repro this, but I'm guessing that you're correct that it's a pixel format issue. The default is BGR24 but, when the camera is configured for MJPEG, it becomes BGRA32. Could you try adding `.ToPixelFormat(PixelFormat.BGR_24bpp)` to [line 55 here](https://github.com/microsoft/psi-samples/blob/main/Samples/LinuxWebcamWithAudioSample/MainWindow.cs#L55)?

dhgarcia, 2020-09-26:
This `.ToPixelFormat(PixelFormat.BGR_24bpp)` indeed changes it from BGRA32 to BGR24 but the frames in the [LinuxWebcamWithAudioSample](https://github.com/Microsoft/psi-samples/tree/main/Samples/LinuxWebcamWithAudioSample) still show in BGR color.
[image]

Which I guess makes total sense since they are BGR.
Only if I add a transformation (maybe there's a better way to do this?) can I get the frames on RGB.
```csharp
var rgb_ImageData = image.Resource.ReadBytes(length);

for (int i = 0; i < length; i++)
{
    if (i % 3 == 0)
    {
        rgb_ImageData[i] = this.imageData[i + 2];       // R < B
        rgb_ImageData[i + 2] = this.imageData[i];       // B < R
     }
}

return new Pixbuf(rgb_ImageData, false, 8, image.Resource.Width, image.Resource.Height, image.Resource.Stride);
```
[image]

AshleyF, 2020-09-28:
I've added a work item to investigate adding an RGB24 `PixelFormat` that could be used with the `ToPixelFormat(...)` operator. Hopefully your workaround is enough to get you going for now.

It's very strange that I don't repro this on my Ubuntu machine. BGR24 bytes displayed in a new `PixBuf` looks normal. At any rate, we should add support for easy RGB conversion. Thanks for raising this!

[image]