microsoft/psi #91: Issue getting past creating a .NET Core console app"1. A simple \psi application" step in "Brief Introduction" on Linux

BrandonThaiTran, 2020-09-23:
Hello,

I am new to PSI so thank you for your patience. 

I am having trouble getting past creating a .NET Core console app "1. A simple \psi application" step in "Brief Introduction" on Linux.

There is no Visual Studios on Linux, only Visual Studios Code, so I cannot do File -> New Project -> Visual C# -> Console App (.NET Core)

Any help would be great!

AshleyF, 2020-09-23:
To create the Brief Introduction app and add the `Microsoft.Psi.Runtime` NuGet on Linux, try this:

```bash
dotnet new console -n BriefIntro && cd BriefIntro
dotnet add package Microsoft.Psi.Runtime --v 0.13.32.2-beta
```

Similarly, when you get to the sample using `WaveFileAudioSource` and `AcousticFeaturesExtractor`, you'll need to `dotnet add package Microsoft.Psi.Audio --v 0.13.32.2-beta`.

This should get you through everything in the Brief Introduction with the exception of section _5. Visualization_ discussing PsiStudio which requires Windows. You might also be interested in the wiki topic on [Bridging to Python](https://github.com/microsoft/psi/wiki/Python-Interop-Walkthrough) which demonstrates using Python for potential ad hoc visualization on Linux.

Have fun!

BrandonThaiTran, 2020-09-24:
Thank you so much! I got the two commands working but I will see how the rest of the intro goes. One thing, the second command only worked when I did `-v` instead of `--v`

closed 2021-01-09 by AshleyF