microsoft/psi #95: BadImageException

kahlilfitz, 2020-10-15:
I followed all of the instructions on the previously closed [issue #40 ](https://github.com/microsoft/psi/issues/40)

I'm still getting the same runtime exception:

```
System.BadImageFormatException
  HResult=0x8007000B
  Message=Could not load file or assembly 'Microsoft.Psi.RealSense_Interop.Windows.x64, Version=0.13.38.2, Culture=neutral, PublicKeyToken=null'. An attempt was made to load a program with an incorrect format.
  Source=Microsoft.Psi.RealSense.Windows.x64
  StackTrace:
   at Microsoft.Psi.RealSense.Windows.RealSenseSensor.Start(Action`1 notifyCompletionTime) in C:\Users\Kahlil Fitzgerald\source\repos\psi\Sources\RealSense\Microsoft.Psi.RealSense.Windows.x64\RealSenseSensor.cs:line 65
   at Microsoft.Psi.Executive.PipelineElement.<Activate>b__46_0()
   at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action, SchedulerContext context)
   at Microsoft.Psi.Scheduling.Scheduler.Run(Object workItem)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()

```

Do you have any other advice than that which was previously mentioned?

chitsaw, 2020-10-20:
The only other thing I can think to check, assuming you've already checked all your project settings have a Platform target of x64 (per the other issue), do you have the RealSense SDK bin folder (C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64) somewhere in your PATH environment variable?

kahlilfitz, 2020-10-20:
Thanks for the feedback, I ended up using the intel RealSense nugget and making a generator class:
[Here's the code](https://github.com/kahlilfitz/realsense-psi-demo)

chitsaw, 2020-10-20:
Glad you found a workaround, and thanks also for sharing your code.

closed 2020-10-20 by chitsaw