microsoft/psi #101: Is there any example or materials on using psi on a NAO robot?

zhaozj89, 2020-12-15:
Hi, thanks for this work. I saw in some videos you show results with the NAO robot. I would like to inquire where should I start if I want to use psi on a NAO robot. Thanks.

AshleyF, 2020-12-18:
Hi Zhenjie, while the Nao videos you're mentioning are likely of prototype systems we've built before \psi, we _have_ experimented also with writing \psi components for interfacing with the Nao/Pepper robots. We will consider publishing these components (or a sample illustrating how to connect to Nao/Pepper) in a future release, as they evolve and reach a more mature stage. For now, I would suggest one of two options:

1) Use the [ROS bridge](https://github.com/microsoft/psi/wiki/ROS-Integration) along with the [NAO package for ROS](http://wiki.ros.org/nao/Tutorials/Installation). You may notice, in fact, that some of the `naoqi_bridge_msgs` ROS message types [are already defined in the ROS bridge](https://github.com/microsoft/psi/blob/master/Sources/Integrations/ROS/Microsoft.Psi.ROS/RosMessageTypes.fs).
2) Use the [Python bridge](https://github.com/microsoft/psi/wiki/Python-Interop-Walkthrough) along with the [NAO package for Python](https://pypi.org/project/qi/). In particular, look at how to [transport \psi messages over ZeroMQ](https://github.com/microsoft/psi/wiki/Python-Interop-Walkthrough#transporting-over-message-queue), how to [receive them in Python](https://github.com/microsoft/psi/wiki/Python-Interop-Walkthrough#receiving-information-from-python-in-psi) and visa versa.

Both options will work (we've used both to work with NAO and Pepper). The Python option may be easier to get working in a Windows-only environment while the ROS option may be easier to bring up on Linux, for the ROS node itself at least.

We'll keep this issue open and keep you posted if/when there is any progress on releasing \psi components for NAO.

Thanks!

zhaozj89, 2020-12-19:
Thanks for the informative reply. I think I can already start with your suggestions. Also, looking forward to your release of \psi components for NAO.