microsoft/psi #102: Open encoded image stream from store

LeebHelmut, 2020-12-22:
Hello,

we create and save an image stream with encode image to reduce the size of the stream:  
```csharp 
azureKinect.ColorImage.EncodeJpeg(90, DeliveryPolicy.LatestMessage).Write("Sources.ColorImage", store);
```
Afterwards we want to read this stream from the store. We tried the following lines of code: 
```csharp 
//Version1:
var colorImage = store.OpenStream<Shared<Microsoft.Psi.Imaging.Image>>("ColorImage");
//Version2:
var colorImage = store.OpenStream<Shared<Microsoft.Psi.Imaging.EncodeImage>>("ColorImage");
```
Both didn't work.
Is there a way to load the data from the store?

Thanks!

danbohus, 2020-12-22:
It seems you are writing it under the name "Sources.ColorImage". When you open a stream you need to specify the type and name under which the stream was persisted in the store. The type is indeed `Shared<EncodedImage>`, but the name should also match, i.e. it should be "Sources.ColorImage" rather than just "ColorImage". So something like your version 2, with the right name, should work: 

```csharp
var encodedColorImage = store.OpenStream<Shared<Microsoft.Psi.Imaging.EncodedImage>>("Sources.ColorImage");
```

Btw, this will be an encoded (jpeg) image stream, which you can decode by applying the `Decode()` stream operator to get a stream of `Shared<Image>`, e.g.:

```csharp
var colorImage = encodedColorImage.Decode();
```
or
```csharp
var colorImage = encodedColorImage.Decode(DeliveryPolicy.LatestMessage);
```

Hope this helps!

LeebHelmut, 2020-12-23:
Thanks for your very fast reply. 

We copied the wrong name of the stream into the issue, but we used the correct one. 
We figured out what the problem was. 
Unfortunately we did not add the NuGet package Microsoft.Psi.Imaging to the project that reads the stream. So I got an exception, that the framework was not able to deserialize the EncodedImage.

Thanks anyway and excuse my error.

danbohus, 2020-12-23:
I see. No worries, and I'm glad you found the problem.

closed 2020-12-23 by danbohus