microsoft/psi #104: PsiStore Importer Missing Data Edgecase

xiangzhi, 2021-01-14:
I found an edge case where data from a PsiStore won't be imported under certain specific scenarios. I think it might have to do with streams with only a single message (e.g. DepthCalibration Information) among multiple streams.

Here's an example PsiStore: [example PsiStore](https://www.dropbox.com/sh/6gjph34sj59hkl1/AABnpustXhNO_LH4tfdzOfada?dl=0)

In the PsiStore, there is a stream `k2d1.depth-calibration` that has a single message. However, if you run the following code:
```
            using (var p = Pipeline.Create(enableDiagnostics: true))
            {
                var input = PsiStore.Open(p, "calibration-recording", @"C:\Data\Broken\broken-recording");
                var calibration = input.OpenStream<IDepthDeviceCalibrationInfo>("k2d1.depth-calibration");
                //var calibration2 = input.OpenStream<IDepthDeviceCalibrationInfo>("azure1.depth-calibration");
                calibration.Do((m,e) => Console.WriteLine("Hello"));
                p.Run();
            }
```
The console never prints `Hello`. However, when I uncomment `var calibration2`, it will work. I sort of trace it back to the way indexing of PsiStores are done and it picking the wrong starting position, but I lack the knowledge on the inner working to know exactly what happened. Please let me know if this just a quirk on my computer or it's reproducible.

chitsaw, 2021-01-14:
It's reproducible. Investigating...

chitsaw, 2021-01-15:
Yep, it's a bug. Thanks for reporting this.

closed 2021-01-15 by chitsaw

xiangzhi, 2021-01-15:
Awesome, thanks for the speedy fix!

PR #106 (2021-01-15): Fixed store seek bug on index boundaries
Fixes #104