microsoft/psi #107: InverseOfClosedForm in CameraIntrinsics.cs cannot handle intrinsics of lower dimensions.

xiangzhi, 2021-01-27:
I'm trying to visualizer depthImage from Kinect 2 using the newly implemented visualizers `(Shared<DepthImage>, ICameraIntrinsics, CoordinateSystem)`. It throws the following exception:
```
System.ArgumentOutOfRangeException
  HResult=0x80131502
  Message=Specified argument was out of the range of valid values.
Parameter name: index
  Source=MathNet.Numerics
  StackTrace:
   at MathNet.Numerics.LinearAlgebra.Storage.VectorStorage`1.ValidateRange(Int32 index)
   at MathNet.Numerics.LinearAlgebra.Vector`1.get_Item(Int32 index)
   at Microsoft.Psi.Calibration.CameraIntrinsics.InverseOfClosedForm(Point2D inputPt, Point2D& outputPt) in C:\Users\Zhi\Source\Repos\CMU-TBD\psi\Sources\Calibration\Microsoft.Psi.Calibration\CameraIntrinsics.cs:line 159
```
The cause is that the camera intrinsic of a Kinect 2 only has 2 radial distortion coefficients(`k`) instead of 6. It might able to be solve by doing some checking and initialize the remaining `k`s to be 0.

sandrist, 2021-01-27:
Thanks Zhi, sounds like we need an extra check inside the `InverseOfClosedForm` method to not assume 6 radial distortion coefficients. I'll collect some Kinect 2 data and debug.

xiangzhi, 2021-01-28:
Awesome, Thanks Sean!

sandrist, 2021-01-28:
Ok, so it seems like the right fix is indeed setting the remaining k coefficients to 0 if they don't exist. I tested on my end and that works. Where were you seeing a denominator of 0? The only denominator I'm seeing is:
```
double h = 1 + k4 * radiusSq + k5 * radiusSqSq + k6 * radiusSq * radiusSqSq;
```
which should be simply equal to 1 if k4, k5, and k6 are 0.

I can check in a quick bug fix, but let me know if I'm missing something.

xiangzhi, 2021-01-29:
You are right. I wasn't really paying attention on my end and actually didn't recompile my PsiStudio correctly and used the old code.

PR #109 (2021-02-04): bug fixes to camera intrinsics and depth camera view as point cloud viz
- The constructor for `CameraIntrinsics` now checks for a valid 3x3 transform matrix. Radial distortion coefficients are initialized as a vector of length 6, copying over however many are passed to the constructor. Same for tangential distortion coefficients (but length=2).
Fixes #107 
- Fixed a small bug in `SpatialDepthCameraViewAsPointCloudVisualizationObject` to not allow the Point Cloud Sparsity property to take any value lower than 1.
closed 2021-02-04 by sandrist