microsoft/psi #108: Behavior of Greedy Interpolators on streams with one message.

xiangzhi, 2021-02-03:
This might be another edge use case that is more suited to be implement on our own. 

As part of my study, I create a component (`StreamValidator`) produces a message when all attached streams have published at least once. It is going to be used to make sure the 3 x AzureKinects and 1 x  Kinect2 are running before the study. 
https://github.com/CMU-TBD/psi/blob/tbd/visual-pipeline/main/TBD/TBD.Psi.StudyComponents.x64/StreamValidator.cs

Since I have a `.First()` operator for each stream, there will only be one message and any reproducible interpolators won't work and I'm force to use the `Fuse()` component. When running, I notice it doesn't work.

I traced the issue back to the situation where the primary stream receive a message but at least one secondary stream has not receive any messages. In that scenario, not messages will ever be passed on. Here's a simple example:
```
           using (var p = Pipeline.Create())
            {
                var stream1 = Generators.Range(p, 1, 1, TimeSpan.FromSeconds(1));
                var stream2 = Generators.Range(p, 10, 1, TimeSpan.FromSeconds(1));
                var stream3 = Generators.Range(p, 100, 1, TimeSpan.FromSeconds(1));
                var keepAlive = Generators.Range(p, 5, 2, TimeSpan.FromSeconds(1));

                var fuser = new Fuse<int, int, int, int>(
                    p,
                    Available.Nearest<int>(TimeSpan.MaxValue),
                    (p, s) => p,
                    2);

                stream1.PipeTo(fuser.InPrimary);
                stream2.PipeTo(fuser.InSecondaries[0]);
                stream3.PipeTo(fuser.InSecondaries[1]);

                fuser.Do(m =>
                {
                    Console.WriteLine(m);
                });
                p.Run();
            }
```
In the example above, there is no guarantee whether the primary or secondary message will arrive first. If the primary arrives first, then no message will ever be published. If you add a delay:
```
stream1.Delay(TimeSpan.FromMilliseconds(10)).PipeTo(fuser.InPrimary);
```
It will work and print out `1`. 

I cannot figure out if this is intended behavior or not.  If yes, maybe there is a space for a greedy interpolator that waits for at least one message on all streams.  I'm also open to other ways of achieving what I want to do.

sandrist, 2021-02-04:
Looking very quickly at your `StreamValidator` component, I wonder if you can accomplish what you want a bit more simply with a basic component (not a subpipeline) and without using Fuse at all. Your component could instantiate any number of receivers, and the receiver method would simply update local state on whether or not a message had been received on that input yet. If the bookkeeping reflects that a message has been received on all incoming streams, you could post a message to the Out emitter (and from that point on the component wouldn't do anything I suppose). Does that make sense? Am I missing any required functionality?

As for what the expected behavior should be with your simple Fuse example, I need to think about it a bit more. I'll try running that code in a few different ways and get back to you.

xiangzhi, 2021-02-05:
Thanks Sean! Yeah, that's a way simpler design and it would work. I guess I over complicated it.

sandrist, 2021-02-06:
Hi Zhi, so in thinking more about your question about the `Fuse` component with a greedy interpolator, the behavior you were seeing should in fact be expected.

The key point is that *it's the primary stream that drives outputs*. When the component receives a message from the primary stream, it asks, "have I received anything on my secondary streams that I can pair with?". If the answer is "yes", it emits a result. Otherwise it doesn't emit anything. The next time a message is received on the primary stream, it checks again. The secondary streams do not drive the emitting of results.

This behavior makes more sense when the streams are dense, but can result in some of these odd situations when you have streams of only one message. When multiple incoming messages have the same originating time, it's fairly random which order the Scheduler will choose to deliver those messages in.

sandrist, 2021-02-06:
Also, there might be an even simpler solution to your problem than the one I described, involving a simple use of the `Merge` operator with a `Process`. Here's the idea applied to your three streams:

```
var sourcesReceived = new HashSet<int>(3);
Operators.Merge(new List<IProducer<int>>() { stream1, stream2, stream3 })
    .Process<Message<int>, bool>((m, env, emitter) =>
    {
        sourcesReceived.Add(m.SourceId);
        if (sourcesReceived.Count == 3)
        {
            emitter.Post(true, env.OriginatingTime);
        }
    })
    .Do(_ => Console.WriteLine("All sources received"));
```

xiangzhi, 2021-02-10:
Thanks for clarification! I think what tripped me up about this was when there are multiple secondary streams and only when one of them is missing a value. I was sort of expecting some kind of incomplete output, but thinking more about it, the incomplete output doesn't really make sense in the context of /psi and its streams too. I'll just have to be mindful of this type of situation in the future :) 

Awesome, the `Merge` solution is a lot more cleaner than my multiple receiver implementation. Thanks!

closed 2021-02-10 by xiangzhi