microsoft/psi #110: Error when trying to use MediaSource class

kahlilfitz, 2021-02-23:
I would like to load a video and use the images as a source for PSI.  I am attempting to do this with the MediaSource class thusly:
` bool fileExists System.IO.File.Exists(location);
  var videoSource = new MediaSource(pipeline, location);`

where `location` is string to the full path of a video file.  I get the following exception:
`System.IO.FileNotFoundException
  HResult=0x80070002
  Message=Could not load file or assembly 'SharpDX.MediaFoundation, Version=4.2.0.0, Culture=neutral, PublicKeyToken=b4dcf0f35e5521f1'. The system cannot find the file specified.
  Source=Microsoft.Psi.Media.Windows.x64
  StackTrace:
   at Microsoft.Psi.Media.MediaSource.InitializeMediaPipeline()
   at Microsoft.Psi.Media.MediaSource..ctor(Pipeline pipeline, Stream input, Nullable`1 startTime, Boolean dropOutOfOrderPackets)
   at Microsoft.Psi.Media.MediaSource..ctor(Pipeline pipeline, String filename, Boolean dropOutOfOrderPackets)
   at DeepEngagementFramework.FrameGenerator.SetupSourceFromFileLocation(Pipeline pipeline, String location) in 
...`

the value of `fileExists` is true and it is clear from the stack trace that it is trying to use a string that is not a file location.  I believe I have used the MediaSource constructor correctly

chitsaw, 2021-02-23:
From the exception text, it looks as though there is a problem loading the SharpDX.MediaFoundation library (which MediaSource depends on), or one of its dependencies. Can you check in the first instance that the SharpDX.*.dll files exist in the bin output folder of your project?

kahlilfitz, 2021-02-24:
I was missing the SharpDX.MediaFoundation nuget package.  Added it and we're good.  Thanks!

closed 2021-02-24 by kahlilfitz