microsoft/psi #111: Unable to open datastore

paung23, 2021-02-25:
In our program, we have a part that (1) writes to a Psi datastore and (2) reads back from the datastore. First, we run the program to write to the datastore. And then on another run, the program reads back from that datastore. Whenever the program tries to open the datastore to read, the following exception has been thrown: System.ObjectDisposedException "Cannot access a closed file".

[image]

The line of code that gets the exception:
`Microsoft.Psi.Data.Importer storeIn = Store.Open(pipeline, "psiAssist", storeDir + "/psiAssist.0000");`

And here is the stacktrace:
>   at System.IO.FileStream.Flush(Boolean flushToDisk)
   at System.IO.FileStream.get_SafeFileHandle()
   at System.IO.MemoryMappedFiles.MemoryMappedView.CreateView(SafeMemoryMappedFileHandle memMappedFileHandle, MemoryMappedFileAccess access, Int64 requestedOffset, Int64 requestedSize)
   at System.IO.MemoryMappedFiles.MemoryMappedFile.CreateViewAccessor(Int64 offset, Int64 size, MemoryMappedFileAccess access)
   at Microsoft.Psi.Persistence.InfiniteFileReader.LoadNextExtent()
   at Microsoft.Psi.Persistence.InfiniteFileReader.MoveNext()
   at Microsoft.Psi.Persistence.MetadataCache.Update()
   at Microsoft.Psi.Persistence.StoreReader..ctor(String name, String path, Action`2 metadataUpdateHandler, Boolean autoOpenAllStreams)
   at Microsoft.Psi.Data.Importer..ctor(Pipeline pipeline, String name, String path)
   at Microsoft.Psi.Store.Open(Pipeline pipeline, String name, String rootPath)
   at AlternativeMainCore.Program.Main(String[] args) in /Users/phyo/Desktop/CPS 490/psiAssistiveAgent/AlternativeMainCore/Program.cs:line 39

Any help would be appreciated. Thank you!

sandrist, 2021-02-27:
Not sure exactly what's going on here, but I have a couple questions that might help us narrow things down and reproduce the issue:
1. Have you tried this with the latest release? It looks like you might be using code from an earlier release (`Importer` is now `PsiImporter` and `Store` is now `PsiStore`). It's possible that you might still have the issue after updating to the latest code, but let us know either way.
2. What environment are you developing in? (Linux? Windows?)

paung23, 2021-03-11:
Thank you for your reply! To answer your questions:
1. We were previously using 0.11.82.2-beta version, but after changing it to the latest release, the issue was solved. However, the only caveat is that it seems like the data reading part is going on an infinite loop (with the latest version of Psi), which is not really a big deal for the purpose of our project.
2. We were running it on Mac OS when we got the data store issue with the 0.11.82.2-beta version. This same version works without any problem on Windows. The latest release is also giving the infinite loop only on Mac OS, not on Windows.

In any case, thank you for your help again! We have somehow got our issue solved, so we are closing this.

closed 2021-03-11 by paung23